cimryan/teslausb #5: Tested! Plus improvement suggestions/asks

rtgoodwin, 2018-10-12:
Not sure if you've had a chance to test on an actual car, but I verified it this morning on a Model 3, v39.7! And great jump on making the setup script; I was going to do that but didn't get to it yet. 👏 

Created the Pi Zero W + 64GB MicroSD setup on a Macbook with Mojave; network share is on a USB drive on a Mac Mini High Sierra. I didn't use the full `df` size, chose 40GB instead. Some notes...

1) It takes about 45 seconds to become available, sometimes less. (Maybe fsck from dirty shutdown?)
2) I didn't use your setup script, but it looks to be right. I may redo things from scratch just to help test that for you. Especially if you end up making the changes below before I do. :) 
3) The requisite improvements asks below; I might try to do them myself but it may take a couple of days to find time (and re-google-learn my bash scripting):

* **Suggest changing the hostname** in `hosts` (127.0.0.1 entry) and `hostname`. Some folks have another raspberrypi on the network and *might* get confused when they try to reconnect to `raspberrypi.local` and wonder where all their changes went. (Like *this* somebody 😉) Suggest "teslapi" or similar. Simple `sed` exercise likely. 
* **Optionally change mount name in vfat mount options**: I believe you can use the name options to specify something so it doesn't show as "NO NAME" when mounted on a computer. Definitely personal choice. 
* Not all hosts are discoverable by name; for example mine isn't. So the script lines that find the share server IP by resolving the hostname won't work for me. Suggest checking the archive name value to see if it's an IP to bypass that bit if needed. 
* I had to either set the `cifs` version option to `1.0`, or remove it (so it autonegotiates). The default of `3` didn't seem to work for me. I did also try setting the Windows compatibility option on the Mac share. Mac CIFS/SMB is often iffy anyway, so there's probably a "most compatible" setting, and it may change after Mojave even. I'm not going to update the share server to Mojave for a bit though, so, fine for now. 

I also changed the archive script to copy `....TeslaCam/*` instead of `....TeslaCam/saved*`, since I've seen at least one other pattern in there. (`recent-` files) This may not be the smart thing to do, not sure if it messes with the rolling recording, but if the goal is to always archive everything (which makes sense as you obviously won't be viewing in the car...), it seems to make sense. I'll have to do a solid hour drive at some point to validate if the `recent-` get compiled down to a single file (which would be a reason to keep them around); just not sure that they do.

cimryan, 2018-10-12:
I've split out the suggestions into separate issues so that they can be tackled one by one.

My read of the initial scripts was that they were intended to just copy the clips that one had elected to save. Do you think that intention should be reconsidered? If so, I'd like to hear more about these "recent-" files so we can decide whether to always copy them or to somehow make that a configurable option.

cimryan, 2018-10-14:
I agree that the recent files seem to represent the rolling buffer. Copying those files off would copy the last ten minutes before the car arrived back home. I don't think those last ten minutes are likely to be more interesting than any other files from the drive, so I don't think they're worth copying without a human directing the system to save the video for them.

rtgoodwin, 2018-10-15:
I'm ok with that. In fact I could even see it go another way and just have the script have an optional loop where it just copies the rolling buffer out to a backup directory. For folks that buy bigger cards to be safe, there is a TON of extra room. So during the archive step, you could just copy the completed recents out to MONTH_DAY folder and boom, less worry about saving. 

Closing this one out, that can be a separate feature request. I want to punch in an optional section to use Pushover to notify your phone when copying is complete. Should just be a few lines (to grab config values and store them, and then a curl call). Just need time... :)

closed 2018-10-15 by rtgoodwin