cimryan/teslausb #8: Suggest checking the archive name value to see if it's an IP to bypass that bit if needed.

cimryan, 2018-10-12:
> Not all hosts are discoverable by name; for example mine isn't. So the script lines that find the share server IP by resolving the hostname won't work for me. 

From https://github.com/cimryan/teslausb/issues/5

cimryan, 2018-10-12:
I think the archiveservername can be specified as an IP address when using the setup-teslausb script. If an IP address is specified that address will be written into /etc/fstab and /root/bin/archiveloop.

Does that address this issue?

cimryan, 2018-10-13:
That line will set archiveserverip to the value of $archiveserver if $archiveserver contains an IP address.

closed 2018-10-13 by cimryan

rtgoodwin, 2018-10-16:
With current code, if an IP is set as the archivename, the fstab is written out as `///ShareName`, so it does seem like it's eating the IP address somehow.