cimryan/teslausb #9: Better initial support for Mac CIFS/SMB

cimryan, 2018-10-12:
> I had to either set the cifs version option to 1.0, or remove it (so it autonegotiates). The default of 3 didn't seem to work for me. I did also try setting the Windows compatibility option on the Mac share. Mac CIFS/SMB is often iffy anyway, so there's probably a "most compatible" setting, and it may change after Mojave even. I'm not going to update the share server to Mojave for a bit though, so, fine for now.

From https://github.com/cimryan/teslausb/issues/5

rtgoodwin, 2018-10-13:
I'm honestly not 100% sure on this one. 

Per https://help.apple.com/deployment/macos/#/ior96b240d12 High Sierra should be version 3.0. Leaving off the version actually makes it default to v1 per https://www.systutorials.com/docs/linux/man/8-mount.cifs/#lbAE so presumably it's more compatible and/or didn't require some alternate setting or config I was missing on the Mac share. 

Ultimately, I just discovered it by reading the archive mount log; it's a silent failure right now unless you know to look there. So I could counter suggest my own suggestion with perhaps doing a test mount during setup if it fails then provide a message about changing the version. (Or if we're feeling particularly helpful, offer to try again with the version not set.)

cimryan, 2018-10-16:
The test mount during the setup process seems like a good idea. Right now the scripts are hosted in a windows_archive directory. Maybe we should split out another directory for Mac hosting.

rtgoodwin, 2018-10-16:
Works for me. It definitely won't mount with v3, tested again today during
the Pushover development.