cimryan/teslausb #11: Archive to SFTP

cimryan, 2018-10-14:
Maybe we'll do this if people express interest.

Gocnak, 2018-10-20:
I think this could be simplified by using [rsync](https://download.samba.org/pub/rsync/rsync.html) (or could even be another archival method using it).

It would be as simple as running `rsync -au $ARCHIVE_DIR $FTP_USER@$FTP_SERVER:$FTP_DIR` which would copy the contents of the `$ARCHIVE_DIR` directory (in this case, the `mnt/cam/TeslaCam/saved/` folder, could also tie in with #37 maybe?) to `$FTP_DIR` on another machine, over ssh.

I will use something like this with another storage pi on my network, with a generated ssh keypair from the teslausb pi as authentication. Once I get it working I can open a PR with it.

Gocnak, 2018-10-21:
I'm currently hitting a few snags with my implementation. I'm using `campercent=100` on a 64GB card, and it creates a ~53GB CAM mount just fine. I can put a TeslaCam folder on it and put files in that folder on Windows, but now the problem comes from reading that drive on the teslausb pi. For some reason, the `/mnt/cam/` folder is empty, despite having the `TeslaCam` folder (and files in it) that persists through ejecting and plugging back in the teslausb pi. Is my script looking at the wrong spot? I know the traditional method is to move the files over from the cam mount to the archive mount, which is mounted to a shared drive. But that isn't needed for rsync, since it can just copy directly from the mount folder.

Another issue is that it also keeps disconnecting even though the archive server is reachable. Individual, repeated calls to `ping` to the ftp/rsync server return the packet just fine, so for some reason it can't reach it, unmounts the USB, remounts the USB, and tries to archive again. It's bizarre.

Gocnak, 2018-10-21:
So I did some further digging and was able to figure out the first issue! It was due to the connection to the ftp server requiring accepting the authenticity of the FTP server's host key (needed to type 'yes'), so I'll be updating the instructions to help ensure people don't have this bizarre issue.

The other two issues remain though, and other than the weird disconnection issue causing the teslausb pi to just spam my computer as if I was plugging in and out a USB drive every 5-20 seconds, it should be ready for merge.

cimryan, 2018-10-21:
I see the malfunctioning device warning, too. It's seemed benign, so I haven' investigated it.

To figure out why the process is deciding that the archive server is unreachable try changing line 14 archiveloop to capture the output of the ping command. Maybe something like this:

```
ping -q -w 1 -c 1 "$ARCHIVE_HOST_NAME" > /tmp/ping.log 2>&1 || reachable=false
```
If you're parked so that the Pi is barely connected to the wireless network I could imagine that this test would flip flop. It is a pretty simple heuristic; perhaps it could be improved to calculate some percentage of successful ping results?

cimryan, 2018-10-21:
When you're ready to merge please submit a pull request.

Gocnak, 2018-10-21:
> If you're parked so that the Pi is barely connected to the wireless network I could imagine that this test would flip flop. It is a pretty simple heuristic; perhaps it could be improved to calculate some percentage of successful ping results?

Interestingly enough it wasn't in the car when it was happening, but rather right at my PC, in really good range of the access point. I will try that command though and let you know!

Perhaps running the 10-try retry on the connection method inside archiveloop could be worth doing?

Gocnak, 2018-10-21:
So, I changed the ping to use the output log and couldn't find anything out of the ordinary. [Introducing the 10-step retry to archiveloop](https://github.com/Gocnak/teslausb/commit/f218c24f4d86d0d76f3d30e805c5ef348d2d77ab) has worked. I'm not sure if this causes any unintended behavior, since the `wait_for_archive_to_be_unreachable` function should be sure that the archive is unreachable, right? So if all 10 pings fail, it would be pretty safe to assume the car/pi cannot reach the archive server.

cimryan, 2018-10-23:
The retry makes sense.

I think we should also add retries for individual file copies.

I expect to be able to regression test your branch on Wednesday evening, if nobody beats me to it.

Gocnak, 2018-10-23:
Alright, I'll open another PR with these changes added!

cimryan, 2018-10-24:
I'm merging this in u/gocnak/rsync_archive.

cimryan, 2018-10-25:
This has been merged. I added an "Expermintal" disclaimer in the main readme. I verified that the merge didn't introduce regressions in the existing CIFS archive scenario but didn't verify that the rsync archiving worked. There are a bunch of outstanding pull requests that I'm going to prioritize over verifying the rsync functionality, myself, so if someone reading this is inclined to try it out please let us know if it works.

rtanaka, 2018-10-25:
@Gocnak - Thanks for laying the foundation for me. I haven't tested the actual rsync part but I can verify that the basic structure works since I followed the exact same structure for the rclone version. The only difference is the actual rsync command and those are pretty straightforward. It's worth noting that we can't keep adding more methods like this since it makes the code a little messy but hopefully our 3 methods cover the majority of the use cases now.

Gocnak, 2018-10-26:
@cimryan - I've been using my implementation of it for a while but I do need to reflash with the latest changes. Once I'll do that, I can let you know how it works and we can close the issue.

@rtanaka - No problem! I agree that it could get a little cluttered, especially since this is mostly pure bash, but if the need ever arises, there could be some merit to organize it with some sort of global config or script or something. The idea would be to set the config beforehand, run the setup script, and the archive loop will properly archive to the desired outputs a user wants.

cimryan, 2018-10-27:
I'm going to assume this works until we hear otherwise.

closed 2018-10-27 by cimryan