cimryan/teslausb #12: Archive to Google Drive

cimryan, 2018-10-14:
Maybe we'll do this if people express interest.

neondemon, 2018-10-19:
+1 interest

Gocnak, 2018-10-21:
Seems like Google has [pretty good documentation on the how-to](https://developers.google.com/drive/api/v3/resumable-upload), but there's a main caveat: to use an access token, you'll need to generate it utilizing their OAuth 2.0 implementation. This deals with a client ID **and more importantly a client secret**, which means it'll need to be obfuscated somehow if it'll be bundled with the pi.

Is using Python for this a little too far fetched for the project? Seems simple enough to be able to do it from command line but having a script that helps with oauth token refreshes will be immensely helpful, and [they have a nice SDK for Python](https://developers.google.com/api-client-library/python/guide/aaa_oauth) as well.

cimryan, 2018-10-21:
Let's use each language in its appropriate place. Bash is good for launching scripts and programs and operating on files, but is lousy for complex logic. Python is complementary. IMHO.

So, I think the tactical application of Python would likely be appropriate.

rtanaka, 2018-10-24:
[rclone4pi](https://github.com/pageauc/rclone4pi/wiki#how-to-configure-a-remote-storage-service) seems like a good generic solution that would enable a lot of cloud based storage mechanisms. i'm testing it out to sync with gdrive right now.

cimryan, 2018-10-27:
I've made changes to create a 100MB partition after the backingfiles partition. You can develop further off of the u/rtanaka/rclone_sync branch.

cimryan, 2018-10-29:
Merged to master: https://github.com/cimryan/teslausb/pull/75

closed 2018-10-29 by cimryan