cimryan/teslausb #15: Make the music and cam folders accessible on the network

sumitkgarg, 2018-10-14:
This way, rather than adding scripts to the pi to copy/move files to the server, the server can detect when the pi is accessible and pull the files or sync the music. Also, a user can manually look at files without copying files on the server first.

cimryan, 2018-10-14:
With the current state in master the Pi is already accessible over the network via ssh/sftp. Is that good enough?

sumitkgarg, 2018-10-14:
Thanks for your response. 

I did not know that. I will try this way (unfortunately, that will be a couple of weeks) and update this thread.

cimryan, 2018-10-15:
Yes, as long as the Pi is powered. Please open a new issue if you find that your scenario isn't working.

closed 2018-10-15 by cimryan

lairdb, 2018-11-07:
(Admission: hardware is on the way, so I haven't implemented or experimented with anything yet.)  

Would it be reasonable/possible to install samba and make the music drive natively accessible when network-attached?

cimryan, 2018-11-08:
@sumitkgarg the filesystem stored in /backingfiles/music.bin isn't mounted. You'll need to mount it. You should also probably unload the g_mass_storage kernel module before you do so.

cimryan, 2018-11-08:
@lairdb Yes, you could probably make that work with a few script modifications.