cimryan/teslausb #17: VPN and multiple AP connect

shahmeeresmail, 2018-10-15:
It would be great to be able to sync media files from multiple networks in addition to their home network such as at work or via a mobile hotspot.

One way to enable this is to provide support for two things:
1.  Allow user to specify multiple SSIDs to automatically connect to when available.
2. Allow the user to specify parameters for a VPN that would automatically be connected to.

This would allow one to securely sync their media files to their private network whenever they connected to a predefined list of wireless access points.

cimryan, 2018-10-15:
You can already achieve this, with an important caveat:
Multiple networks can be specified while editing the wpa_supplicant.conf file by creating multiple network blocks.
However, this guide currently leaves the Pi insecure: SSH enabled with a default password. That's OK as long as your only connecting to networks that you control, but connecting to a shared network means that the Pi could get exploited. So, if we add instructions for how to connect to multiple networks we should also add instructions for how to lock down SSH on the Pi.

cimryan, 2018-10-16:
I've split the VPN idea off into a separate issue: https://github.com/cimryan/teslausb/issues/21

It'd still be good to have instructions for securing the Pi.

cimryan, 2018-10-25:
Merged. I'd like to see instructions added for securing the Pi before any references are added to the add-wifi.ps1 script in the documentation.

skipfire, 2018-10-25:
I will work on an update to readme.

skipfire, 2018-10-25:
Pull #61 for updating the docs.

cimryan, 2018-10-26:
OK, merged.

closed 2018-10-26 by cimryan