cimryan/teslausb #22: Unable to identify as usb drive on PC

pqi8, 2018-10-16:
32gb SD card

Went through the following steps:

1. Used Etcher on Windows PC to flash Raspbian Stretch Lite onto SD card.
2. Using mini hdmi/keyboard - did not use headless setup
3. Went through all of the steps - Archive server was reachable 
4. Set campercent to 100
5. After reboot - got an error regarding "/mnt/archive" - Something about unable to Mount. Everything 
    else showed OK. 
6. Wasn't sure what to do at this point, so I unplugged the pi and plugged it back into Windows PC
7. Windows showed 2 drives - One as a boot drive with around 40mb total space 20 used and one as 
    29gb empty unformatted drive. The drive was not accessible, so I tried to format as FAT32 and created 
    TeslaCam directory, but I don't think I was in a usable state regardless.  

Had a friend look at it, and it seems like the partitioning might not be giving me enough space to mount the archive? Any tips and what I should try to do next, or if I did something wrong?

Thanks.

rtgoodwin, 2018-10-16:
I'm seeing the same thing, I think the recent change to manage the percentage of drive allocated broke something. Poking around.

rtgoodwin, 2018-10-16:
I ended up wiping the disk before taking more notes, but @cimryan I did notice that when the vfat drive wasn't mounted to `/mnt/cam` it was showing mounted to `loop0`. Not sure what that means TBH. 

I tried recreating it at continuing smaller sizes, down to 40GB, and it still wasn't working. It's a headscratcher. Can't test with current master due to #25.

cimryan, 2018-10-16:
I failed to test the 100% scenarios. I imagine that I broke something, there. I'll look at this. In the meantime, if you'd like to play, try setting a value less than 100 for the campercent, e.g. 90.

pqi8, 2018-10-16:
@cimryan I've tried 75 and 50, I get the same issue. Let me know if you have anything else you'd like me to try.

cimryan, 2018-10-17:
The problem was that the instructions for folks using the keyboard/hdmi setup weren't updated to include removing the command that tells the Pi to expand the /dev/mmcblk0p2 partition to fill the drive on initial boot. I've now updated those instructions, although they really should be expanded and automated, in the future. If you'd like to try again I expect you'll find success. I'm still working on the 100% scenarios.

cimryan, 2018-10-17:
I confirmed that setting campercent=100 works.

lolento, 2018-10-17:
> I confirmed that setting campercent=100 works.

I just tried a 32gb card and 64gb card...getting similar errors.  I've been trying different cards and different percentages pretty much all night but nothing is working.

pqi8, 2018-10-17:
I missed the new line:

"Before you start those steps, ensure that the MicroSD card is in your computer and edit the cmdline.txt file to delete the parameter " init=/usr/lib/raspi-config/init_resize.sh"."

I will apply that first and let you guys know. Thanks.

cimryan, 2018-10-17:
I think this has been addressed by the removal of init_resize

closed 2018-10-17 by cimryan