cimryan/teslausb #23: Missing files.  Unsure how this works...

lolento, 2018-10-16:
So I got this thing setup last night and it seems to work...

I have plug the drive to my Tesla and I see a partial file copied to my TM3 folder in my windows server.

However, this morning I only see that a 5mb file is copy to my shared folder under this name:
saved-front-2018-10-16_01-36.mp4 

And this file is incomplete because it is not playable.

So then, I got to the office today and directly plugged the drive into my laptop.  I don't even see this file on the drive anymore. 

So a couple of questions....(sorry, didn't mean to combine questions):
1.  Is this the correct behavior?  When the readme says "archive" does it mean it will delete the copied over file?  (If so, can I just assume in this case that Tesla didn't record correctly?)
2.  Or does this script require the Pi Zero to be constantly powered on? (Did I shut off the car too quickly?)
3.  Does this script record everything that start with "Recent-...."?  Or only the "Saved-Front-..."?

Thanks!

rtgoodwin, 2018-10-16:
The script moves (not copies) the saved* files currently. If the file isn't 0 bytes, then the dashcam probably didn't finish saving out the file. At least my experience so far has been that a copied file is 'whatever size it was on the flash drive' and a file that didn't finish copying is 0 bytes.

Now, why some of the saved files are too small, I'm not sure. I have a suspicion that if you haven't had 10 minutes of footage recorded yet, the save function isn't perfect.

lolento, 2018-10-16:
Thx, so you have this running already?  How do you find this working out?  

I cannot find any information on how long the usb stay powered after the car is off.  Do you find it working in terms of moving all of the save files without any user intervention?

Additionally, my drive just setup today is already corrupted (as recognized by Tesla)....  Is there a way to prevent this from happening?   I know if I plug this into windows and do a file scan, it will start working again....

cimryan, 2018-10-16:
Moving the files off the drive can take from a few seconds to tens of minutes, depending on the strength of the wireless signal and the number of clips being transferred. You can ensure that all the clips are transferred by turning on the climate control using your phone. This will turn on the USB ports.

cimryan, 2018-10-17:
What I see when the files have been partially copied off the drive is that once the Pi is connected to the network again partially copied files are copied, again, overwriting incompletely copied files.

I'm very interested to hear more about your corrupted drive. Changes to prevent drive corruption were introduced at around 7:41 PDT. Did you set up your Pi before or after that?

lolento, 2018-10-17:
Re: to the corrupted drive....

Very funny behavior, so I plugged into Windows and it says to scan for errors... I didn't do that

BUT!  The Pi still continues to copy files to my Share like normal.  So Windows doesn't recognize the drive and Tesla doesn't recognize it but Pi still does and the videos copied over look fine.

Unfortunately, I did a full re-setup of the Micro SD Card.  Next time this happens again, I'll let you know...

cimryan, 2018-10-17:
OK, sounds like we can close this issue.

closed 2018-10-17 by cimryan