cimryan/teslausb #26: Setup Wifi on boot volume before mounting using update script.

rtgoodwin, 2018-10-16:
I can work on this one, since I want to update the script that sets up USB networking anyway. 

You can put the credentials in /boot/wpa_supplicant.conf before booting the Pi the first time and it will autocopy and configure on first boot. 

If you split the dirs out into Windows and Mac/Linux then that will help even more in this case since we can ensure that someone can edit the USB connect script with a *nix compliant editor. 

Then just run it, eject the card, and boot your Pi into glorious USB and (assuming you did it right) Wifi options right off the bat.

cimryan, 2018-10-16:
That's a neat trick!

rtgoodwin, 2018-10-16:
Wish I came up with it, but I imagine it was lots of Raspi devs weary of having to swap SD cards back and forth ;) 

If we reaaaally wanted to supercharge this thing, we could use Hypriot, which has cloud-init built in. Basically you set all your configs and stage your scripts and then on boot cloud-init will take care of executing it all for you. If this became a baked image, or wanted to fully automate, it's something to consider. :) https://blog.hypriot.com/post/cloud-init-cloud-on-hypriot-x64/

cimryan, 2018-10-19:
Just need to update setup-piForHeadlessConfig.ps1.

closed 2018-10-19 by cimryan