cimryan/teslausb #27: Copy all footage rather than just save*

lolento, 2018-10-17:
Is it possible to just copy over all recorded footage rather than the save*?

cimryan, 2018-10-17:
Yes, it's possible. To make this change find the following line in archive-teslacam-clips:
```
for file_name in "$CAM_MOUNT"/TeslaCam/saved*; do
```
Change it to
```
for file_name in "$CAM_MOUNT"/TeslaCam/*; do
```
We probably won't make this the default behavior. See the discussion in this issue for the reasoning: https://github.com/cimryan/teslausb/issues/5

Does that work for you?

cimryan, 2018-10-17:
I'll close this, for now. Please open another issue if it turns out this model doesn't work for you.

closed 2018-10-17 by cimryan

funat120mph, 2019-01-30:
Thanks for all the work on this project.  I can follow instructions but would not be able to develop this on my own.  I believe I have my TeslaUSB working now.  Received a pushover notification but no files yet.

In any event, how would I go about editing archive-teslacam-clips after initial setup?  I prefer to save archive all files instead of the saved clips.  I'm using gdrive and rclone4pi.

Thank you.