cimryan/teslausb #28: make-root-fs-readonly.sh is very slow to run

firedfly, 2018-10-17:
Is the _apt-get update_ part of make-root-fs-readonly.sh required?  I let that step run for a few minutes and it was estimating it had more than 30 minutes remaining to update the package list.

If this is not essential, it may be worthwhile to remove the package update from the script.

cimryan, 2018-10-17:
Thanks for reporting this experience, firedfly.

I also noticed that the update was very slow, this evening. I was seeing download rates as low as 5kb/s. The process was much faster yesterday, so I'm not sure if this will be a continuing pain point or not.

I'm also not sure of what the potential implications of skipping this step are, or how we might skip some of the data.

Until someone does some research to demonstrate that it's safe to skip this step, or to identify a faster way of achieving the same result, or we identify that this is a continuing problem that we can't tolerate, I'm not inclined to disable the update.

Let's keep this issue open while we gather data.

rtgoodwin, 2018-10-17:
It took about 3-4 minutes for me. The Pi has the bandwidth, but sometimes it's...unstable. I've seen `wget` hang for a good minute or so occasionally. Which jogs my mind about adding some stats to the copy procedure to see just how fast things are getting xferred and set user expectations.

firedfly, 2018-10-18:
The slowness on updating the package index doesn't appear to be common.  I've re-configured a pi a few times and updating of the index runs in an acceptable time.

I looked into the script used as a starting point for `make-root-fs-readonly.sh`.  Based on commit history several packages are removed as they will not work with a read-only file system.  In a few instances, replacement packages are installed.

Based on this information, it seems advisable to keep the script as is.