cimryan/teslausb #30: Getting this error when setting up with 32gb and 64gb cards

lolento, 2018-10-17:
I'm getting this error right before the reboot step.  So after reboot, the CAM drive does not show up on windows.

Error: The location 62535680 is outside of the device /dev/mmcblk0.

pqi8, 2018-10-17:
I'm getting something similar 

Error: The location 62521344 is outside of the device /dev/mmcblk0

firedfly, 2018-10-17:
I'm getting this error with an 8gb card.  It looks like the root partition is taking up the all available space on the card.  When the _backingfiles_ partition is created, it attempts to be created one sector above the last sector on the card.  The _backingfiles_ partition tries to start with sector _15523840_.  Below is the info from parted:

> Model: SD SL08G (sd/mmc)
> Disk /dev/mmcblk0: 15523840s
> Sector size (logical/physical): 512B/512B
> Partition Table: msdos
> Disk Flags:
> 
> Number  Start   End        Size       Type     File system  Flags
>  1      8192s   97889s     89698s     primary  fat32        lba
>  2      98304s  15523839s  15425536s  primary  ext4

I think I know why I ran into this issue.  I did not run the powershell script to configure the sd card in Windows.  Instead, I manually configured the wifi config file and added an _ssh.txt_ file.  Apparently, by default, Raspbian will resize the root partition to fill the drive on the first boot.  This is disabled by the _setup-piForHeadlessConfig.ps1_ script.

@lolento , @pqi8  did you configure the card in Windows?  If so, did you run the powershell script to do the configuration?

firedfly, 2018-10-17:
I just read issue #22 and see that these issues appear to be the same.  @cimryan made comments in #22 that should fix the issue (updated instructions).

lolento, 2018-10-17:
Yep, this is fixed with the updated instructions for HDMI and Keyboard setup to edit the cmdline.txt.

rtgoodwin, 2018-10-17:
Confirmed the error is gone when removing the built in resize from cmdline.txt. However the USB drive still isn't mounting to the host. 

```
root@teslausb:~# mount -a -v
/proc                    : already mounted
/boot                    : already mounted
/                        : ignored
/backingfiles            : already mounted
/mnt/cam                 : ignored
/mnt/archive             : already mounted
/var/log                 : already mounted
/var/tmp                 : already mounted
/tmp                     : already mounted
```
```
root@teslausb:~# ls -la /backingfiles/
total 56235672
drwxr-xr-x  3 root root        4096 Oct 17 15:36 .
drwxr-xr-x 23 root root        4096 Oct 17 15:36 ..
-rw-r--r--  1 root root 57585291264 Nov  3  2016 cam_disk.bin
drwx------  2 root root       16384 Oct 17 15:36 lost+found
```
```
root@teslausb:~# cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=7fd599a5-01  /boot           vfat    defaults,ro          0       2
PARTUUID=7fd599a5-02  /               ext4    defaults,noatime,ro  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

/dev/mmcblk0p3 /backingfiles ext4 auto,rw,noatime 0 2
/backingfiles/cam_disk.bin /mnt/cam vfat noauto,users,umask=000 0 0
//MYIPHERE/TeslaCamFiles /mnt/archive cifs credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0

tmpfs /var/log tmpfs nodev,nosuid 0 0
tmpfs /var/tmp tmpfs nodev,nosuid 0 0
tmpfs /tmp    tmpfs nodev,nosuid 0 0
```
Not sure where the mount logic or file creation logic changed to break this.

rtgoodwin, 2018-10-17:
@lolento @firedfly are you confirming that it's now mounting properly as a USB drive, or just that it doesn't error out?

lolento, 2018-10-17:
> @lolento @firedfly are you confirming that it's now mounting properly as a USB drive, or just that it doesn't error out?

Yes, I redid the setup with a 32gb card and now, on both Windows and Tesla, I can see both "Cam" drive and "Music" drive (each at 16gb)

lolento, 2018-10-17:
I can do that when I get home today.

rtgoodwin, 2018-10-17:
Thx! Mine is a 64GB card btw.

cimryan, 2018-10-17:
It looks like the original issue here is resolved, so I'm moving the bit about the drives not showing up to a new issue.

closed 2018-10-17 by cimryan