cimryan/teslausb #32: Drives aren't showing up on the host after setup

cimryan, 2018-10-17:
the USB drive still isn't mounting to the host.

```
root@teslausb:~# mount -a -v
/proc                    : already mounted
/boot                    : already mounted
/                        : ignored
/backingfiles            : already mounted
/mnt/cam                 : ignored
/mnt/archive             : already mounted
/var/log                 : already mounted
/var/tmp                 : already mounted
/tmp                     : already mounted
```
```
root@teslausb:~# ls -la /backingfiles/
total 56235672
drwxr-xr-x  3 root root        4096 Oct 17 15:36 .
drwxr-xr-x 23 root root        4096 Oct 17 15:36 ..
-rw-r--r--  1 root root 57585291264 Nov  3  2016 cam_disk.bin
drwx------  2 root root       16384 Oct 17 15:36 lost+found
```
```
root@teslausb:~# cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=7fd599a5-01  /boot           vfat    defaults,ro          0       2
PARTUUID=7fd599a5-02  /               ext4    defaults,noatime,ro  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that

/dev/mmcblk0p3 /backingfiles ext4 auto,rw,noatime 0 2
/backingfiles/cam_disk.bin /mnt/cam vfat noauto,users,umask=000 0 0
//MYIPHERE/TeslaCamFiles /mnt/archive cifs credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0

tmpfs /var/log tmpfs nodev,nosuid 0 0
tmpfs /var/tmp tmpfs nodev,nosuid 0 0
tmpfs /tmp    tmpfs nodev,nosuid 0 0
```
Not sure where the mount logic or file creation logic changed to break this.

_Originally posted by @rtgoodwin in https://github.com/cimryan/teslausb/issues/30#issuecomment-430683946_

rtgoodwin, 2018-10-17:
I'll test this in the car later. I've rebooted my machine and don't know why it's not coming up. Would be fine if it's a Mac thing I suppose, but don't know why it would have stopped. It doesn't even come up as a disk with an unrecognizable format or anything. Very odd.

rtgoodwin, 2018-10-17:
Interesting, I guess `/mnt/cam` is ok to be ignored. So I don't know how to further troubleshoot here, may have to do some googling. Tested on another Mac, different OS level, and it's still not showing up as a USB disk.

rtgoodwin, 2018-10-17:
I'm officially (in this case) an idiot. All my testing is in a scenario where the ARCHIVESERVER is always reachable. So, of course the drive gets unmounted from my Mac. (Facepalm)

So this makes this really a dev problem, or if someone wants to inspect the USB share on their machine while the Pi is on the network.  (The latter being a pretty reasonable case, actually.)

I'm not sure how to get around that. 

I suppose you could:

-  remount the USB share `ro` once the archive is reachable. Then at least someone could manually copy off files using their PC if the archive process was just not super successful/consistent. If it's plugged into the car, Tesla _should_ detect it dis/appearing, see it's not writeable, and that'd probably be fine since you likely don't want footage of the inside (or outside) of your garage/parking structure. The downside would be if Tesla does release a method to provide constant power to both dashcam AND the USB ports, but that seems unlikely for the moment. 

- Then update the copy process to test presence of the target `saved-` file and if it `exists && is same size (or date?)` then don't copy, just delete it from the `rw` `/mnt/cam`. 

Thoughts? This probably ends up getting closed and a new issue opened to allow for this scenario if we want it. It seems like it has value.

cimryan, 2018-10-18:
The archive-teslacam-clips script starts with disconnecting the drives from the usb host and ends with connecting them. So, the drives should only be missing from the host while clips are being copied.

However, the script is run with -eu, so if anything goes wrong the whole thing exits. I suspect that you're not seeing the drives get remounted because something is going wrong during an execution of archive-teslacam-clips, and so execution isn't reaching the last line, which reloads g_mass_storage.

Do you see "Connected usb to host." in /tmp/archive-teslacam-clips.log?

rtgoodwin, 2018-10-18:
You nailed it. Stray typo in the local file was causing it to silently exit. Time to wipe and re-setup from master anyway. The good news is I now have remembered a small fraction of the large amount of filesystem knowledge I forgot.

abhinavWOT, 2018-10-18:
I am having an issue with the drives "CAM" and "MUSIC" not showing up on windows/mac. This is the output from the log.

`Thu 18 Oct 01:39:17 UTC 2018
Finished moving clips to archive.
Thu 18 Oct 01:39:17 UTC 2018
Unmounting cam drive...
Thu 18 Oct 01:39:18 UTC 2018
Unmounted cam drive.
Thu 18 Oct 01:39:18 UTC 2018
Connecting usb to host...
Thu 18 Oct 01:39:18 UTC 2018
Connected usb to host.
`

What am I doing wrong?

cimryan, 2018-10-18:
@abhinavWOT will you please share the output from these commands?

```
cat /etc/fstab
```

```
sudo -i
modprobe g_mass_storage
dmesg | tail -n 30
```

abhinavWOT, 2018-10-18:
proc            /proc           proc    defaults          0       0
PARTUUID=7d0a06fe-01  /boot           vfat    defaults,ro          0       2
PARTUUID=7d0a06fe-02  /               ext4    defaults,noatime,ro  0       1
a swapfile is not a swap partition, no line here
 use  dphys-swapfile swap[on|off]  for that

/dev/mmcblk0p3 /backingfiles ext4 auto,rw,noatime 0 2
/backingfiles/cam_disk.bin /mnt/cam vfat noauto,users,umask=000 0 0
/backingfiles/music_disk.bin /mnt/music vfat noauto,users,umask=000 0 0
//192.168.2.73/SailfishCam /mnt/archive cifs username=sailfish,password=teslacam,uid=1000,gid=1000,workgroup=workgroup,vers=1.0,users,auto,user_xattr 0 0

tmpfs /var/log tmpfs nodev,nosuid 0 0
tmpfs /var/tmp tmpfs nodev,nosuid 0 0
tmpfs /tmp    tmpfs nodev,nosuid 0 0

[    9.198155] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    9.220014] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[    9.220401] usbcore: registered new interface driver brcmfmac
[    9.830252] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[    9.831477] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 
[   11.855219] uart-pl011 20201000.serial: no DMA platform data
[   14.591682] NOHZ: local_softirq_pending 40
[   14.949598] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   14.949651] brcmfmac: power management disabled
[   16.028721] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.343238] Bluetooth: Core ver 2.22
[   16.343436] NET: Registered protocol family 31
[   16.343448] Bluetooth: HCI device and connection manager initialized
[   16.343479] Bluetooth: HCI socket layer initialized
[   16.343500] Bluetooth: L2CAP socket layer initialized
[   16.343553] Bluetooth: SCO socket layer initialized
[   16.370758] Bluetooth: HCI UART driver ver 2.3
[   16.370781] Bluetooth: HCI UART protocol H4 registered
[   16.370789] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.376389] Bluetooth: HCI UART protocol Broadcom registered
[   16.961856] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.961874] Bluetooth: BNEP filters: protocol multicast
[   16.961911] Bluetooth: BNEP socket layer initialized
[   23.525229] FS-Cache: Netfs 'cifs' registered for caching
[   23.526036] Key type cifs.spnego registered
[   23.526067] Key type cifs.idmap registered
[   28.095658] udc-core: couldn't find an available UDC - added [g_mass_storage] to list of pending drivers
[  176.883002] NOHZ: local_softirq_pending 40
[  189.623080] random: crng init done
[  189.623104] random: 7 urandom warning(s) missed due to ratelimiting

Noticed this:
`udc-core: couldn't find an available UDC - added [g_mass_storage] to list of pending drivers`

cimryan, 2018-10-18:
@abhinavWOT Which directions did you follow to set up this Pi? You seem to have missed a few key steps.

abhinavWOT, 2018-10-18:
I did follow the ones on the readme page, I changed the share on the fstab though because it wasn't connecting and would say host is down. Is that causing the issue?

cimryan, 2018-10-18:
Did you set up the Pi using a Windows machine, a Mac, or a Linux machine?

abhinavWOT, 2018-10-18:
a mac

cimryan, 2018-10-18:
@abhinavWOT You should wait for this pull request to be merged.

https://github.com/cimryan/teslausb/pull/35

It has more explicit instructions and automates the parts that you're missing.

closed 2018-10-18 by cimryan