cimryan/teslausb #34: With current code, if an IP is set as the archivename, the fstab is written out as `///ShareName`

cimryan, 2018-10-17:
it does seem like it's eating the IP address somehow.

_Originally posted by @rtgoodwin in https://github.com/cimryan/teslausb/issues/8#issuecomment-430364393_

cimryan, 2018-10-18:
@rtgoodwin I'm unable to reproduce this.

What does this say when you run it on your Pi?

```
archiveserver=your.archive.server.Ip.address
echo "$(getent hosts $archiveserver | cut -d' ' -f1)"
```

rtgoodwin, 2018-10-18:
Here you go:

[image]

Same when run as `root`.

cimryan, 2018-10-18:
Do you get that result if you run the command before you run setup-teslausb?

Can you ping the IP address successfully?

cimryan, 2018-10-18:
If you can ping the IP address, what do you get from this?

```
archiveserver=your.archive.server.IP.address
echo "$(ping -c 1 -w 1 $archiveserver 2>/dev/null | head -n 1 | grep -o -e "(\([[:digit:]]\{1,3\}\.\)\{3\}[[:digit:]]\{1,3\})" | tr -d '()')"
```

rtgoodwin, 2018-10-18:
That works for me. (And wifi isn't up, so technically the IP isn't even
reachable.)

```
root@raspberrypi:~# archiveserver=172.16.1.100
root@raspberrypi:~# echo "$(ping -c 1 -w 1 $archiveserver 2>/dev/null |
head -n 1 | grep -o -e "(\([[:digit:]]\{1,3\}\.\)\{3\}[[:digit:]]\{1,3\})"
| tr -d '()')"
172.16.1.100
root@raspberrypi:~#
```

cimryan, 2018-10-18:
How are you getting past check_archive_server_reachable in setup-teslausb ?

rtgoodwin, 2018-10-18:
¯\\_(ツ)_/¯

```
root@raspberrypi:~# ./setup-teslausb
Verifying environment variables...
Verifying that the archive server 172.16.1.100 is reachable...
The archive server is reachable.
Verifying that there is sufficient space available on the MicroSD card...
There is sufficient space available.
/tmp ~
--2018-10-18 11:39:20--  https://raw.githubusercontent.com/cimryan/teslausb/master/windows_archive/create-backingfiles-partition.sh
...
```

cimryan, 2018-10-18:
I've updated the setup-teslausb script to set the ip address by parsing the ping output. Please let me know if this resolves the issue; I've never been able to reproduce the failure you encountered.