cimryan/teslausb #43: WiFi is not connected

ken830, 2018-10-21:
Following the latest instructions. I finish the "Get a shell on the Pi" section and I'm logged in via SSH. But the next sections starts with "Now that you have Wifi up and running....", but my WiFi is not up and running. I believe earlier instructions required manually starting the wlan interface? Now it's supposed to be automatic?

I run ifconfig and wlan0 is up, but doesn't look connected (no IP Addr or anything). 

I checked `/etc/wpa_supplicant/wpa_supplicant.conf `and ssid and password look good.

running `sudo wpa_cli -i wlan0 reconfigure` gives this error:

`Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory`

What am I doing wrong? I got it up and running last week.

shahmeeresmail, 2018-10-21:
Are you trying to connect to a 5ghz AP by any chance?

cimryan, 2018-10-21:
Are you using Windows, Linux, or Mac OS to set up the Pi?

rtgoodwin, 2018-10-21:
I have the same AP and no issues. It sounds like the headless setup script maybe didn't complete the wpa_supplicant.conf file? It should end up looking like this before you reboot:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
  ssid="your_ssid"
  psk="your_wifipass"
  key_mgmt=WPA-PSK
}
```
I think you should be able to put the SD card back into your computer to doublecheck/recreate.

ken830, 2018-10-22:
Just started over again starting with Etcher to re-image the SD card. This time, I used another Windows PC (my SurfaceBook) also running Windows 10. Same exact behavior. This is what I'm seeing:

```
root@raspberrypi:~# ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 172  bytes 16422 (16.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 172  bytes 16422 (16.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.189.173  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::88eb:c086:31ea:c131  prefixlen 64  scopeid 0x20<link>
        ether b6:09:ea:5f:e5:e2  txqueuelen 1000  (Ethernet)
        RX packets 471  bytes 64249 (62.7 KiB)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 485  bytes 67554 (65.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:a9:54:6a  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

This has worked when I set it up last week with the same PC, RPi, SD Card, and WiFi.

ken830, 2018-10-22:
Something is weird with github today. I posted new comment here, and then it says something like I can't post the comment. Then, when I tried to comment again, I saw a duplicate comment. Then, the entire github.com/cimryan/teslausb is inaccessible. Now, it's back and my comment is gone.

rtgoodwin, 2018-10-22:
Yeah I wrote and rewrote a few comments yesterday, I shudder to think how many dupes I'm going to find. 

Here are the lines related to moving the file over from /boot on bootup (among otherthings). See if they give you any hints about what state you're in? 

```
/lib/systemd/system/wpa_supplicant@.service:Alias=multi-user.target.wants/wpa_supplicant@%i.service
/lib/systemd/system/wpa_supplicant@.service:ExecStart=/sbin/wpa_supplicant -c/etc/wpa_supplicant/wpa_supplicant-%I.conf -Dnl80211,wext -i%I
/lib/systemd/system/raspberrypi-net-mods.service:Description=Copy user wpa_supplicant.conf
/lib/systemd/system/raspberrypi-net-mods.service:ConditionPathExists=/boot/wpa_supplicant.conf
/lib/systemd/system/raspberrypi-net-mods.service:ExecStart=/bin/mv /boot/wpa_supplicant.conf /etc/wpa_supplicant/wpa_supplicant.conf
/lib/systemd/system/raspberrypi-net-mods.service:ExecStartPost=/bin/chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf
/lib/systemd/system/wpa_supplicant.service:ExecStart=/sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

```

cimryan, 2018-10-22:
I was able to reproduce this failure, this morning. I'll investigate tonight.

skipfire, 2018-10-22:
I am also having this issue and my 2.4 and 5.0 are on different names and I am using my 2.4 ssid.  As for the conf file contents, I have no clue how to read it through SSH or on my PC as the only thing that I see on my PC is the boot which doesn't have much.

cimryan, 2018-10-23:
The problem has been fixed with commit 7a399dee159e0d263ec1d898218000939f7d0586.

closed 2018-10-23 by cimryan