cimryan/teslausb #47: Device not recognized after reboot

skipfire, 2018-10-23:
Downloaded new files today (October 23rd around 9:30 EDT) and after running setup-teslausb and rebooting it comes up wtih a USB device not recognzied error.  I've tried multiple cables that work for data transfer, using both data and a second on the power to ensure it isn't insufficient power, reflashing and redoing the whole process.  Also has this issue yesterday before the commits last night and this morning.  Attached is the powershell window, hope I scrubbed it all.

[Detail.txt](https://github.com/cimryan/teslausb/files/2506499/Detail.txt)

cimryan, 2018-10-23:
I don't think you're supposed to connect the Pi with more than one cable at a time. But that's probably not the issue.

Can you share the content of /tmp/archiveloop.log ?

skipfire, 2018-10-23:
The second USB cable was after I was already having problems to rule out power.  I'd be happy to share the file, how do I do so?  I am able to get in with SSH by IP, just need to know how to echo the file out.

skipfire, 2018-10-23:
Not a lot here so I pasted the whole thing.

root@teslausb:~# cat /tmp/archiveloop.log
Thu  3 Nov 17:17:03 UTC 2016
Starting...
Thu  3 Nov 17:17:04 UTC 2016
Ensuring music backing file is mounted...
Thu  3 Nov 17:17:04 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:04 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:04 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:05 UTC 2016
Retrying...
Thu  3 Nov 17:17:05 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:06 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:06 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:07 UTC 2016
Retrying...
Thu  3 Nov 17:17:07 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:07 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:07 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:08 UTC 2016
Retrying...
Thu  3 Nov 17:17:08 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:08 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:08 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:09 UTC 2016
Retrying...
Thu  3 Nov 17:17:09 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:09 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:09 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:10 UTC 2016
Retrying...
Thu  3 Nov 17:17:10 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:10 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:10 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:11 UTC 2016
Retrying...
Thu  3 Nov 17:17:11 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:11 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:11 UTC 2016
Sleeping before retry...
Thu  3 Nov 17:17:12 UTC 2016
Retrying...
Thu  3 Nov 17:17:12 UTC 2016
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Thu  3 Nov 17:17:12 UTC 2016
Failed to mount /mnt/music.
Thu  3 Nov 17:17:12 UTC 2016
Sleeping before retry...
Tue 23 Oct 14:45:17 UTC 2018
Retrying...
Tue 23 Oct 14:45:17 UTC 2018
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Tue 23 Oct 14:45:17 UTC 2018
Failed to mount /mnt/music.
Tue 23 Oct 14:45:17 UTC 2018
Sleeping before retry...
Tue 23 Oct 14:45:18 UTC 2018
Retrying...
Tue 23 Oct 14:45:18 UTC 2018
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Tue 23 Oct 14:45:18 UTC 2018
Failed to mount /mnt/music.
Tue 23 Oct 14:45:18 UTC 2018
Sleeping before retry...
Tue 23 Oct 14:45:19 UTC 2018
Retrying...
Tue 23 Oct 14:45:19 UTC 2018
Mounting /mnt/music...
mount: can't find /mnt/music in /etc/fstab
Tue 23 Oct 14:45:19 UTC 2018
Failed to mount /mnt/music.
Tue 23 Oct 14:45:19 UTC 2018
Attempts exhausted.
root@teslausb:~#

skipfire, 2018-10-23:
I was able to execute mount for cam, so I'm guessing the script dies before getting to that mount.  The backing files has cam & lost+found, but no file for music (which I would did not expect).

cimryan, 2018-10-23:
Thanks. I believe that this has been fixed.

Reflash and resetup, or
```
/root/bin/remount_rw
cd /root/bin
rm archiveloop
wget https://raw.githubusercontent.com/cimryan/teslausb/master/windows_archive/archiveloop
chmod +x archiveloop
sed -i "s/archiveserver/sfs06/" archiveloop
reboot
```

skipfire, 2018-10-23:
Tried that and the unrecognized device error went away, but I'm not seeing any drives mounted.  I will try flashing on another SD card in case there is any information you want me to get off this one.

cimryan, 2018-10-23:
Did archiveloop exit? If so, what are the last lines in /tmp/archiveloop.log, now?

If you run 
```
lsmod | grep g_mess_storage
```
Do you see output?

If not, when you run
```
modprobe g_mass_storage
/bin/sleep 3
dmesg | tail -n 30
```
Do you see any interesting errors?

skipfire, 2018-10-23:
No output for the lsmod line, the CAM drive appeared right after the modprobe line.  I don't notice any errors, just warnings.

[  112.212995] NOHZ: local_softirq_pending 42
[  323.393010] NOHZ: local_softirq_pending 02
[  325.993055] random: crng init done
[  325.993075] random: 7 urandom warning(s) missed due to ratelimiting
[  362.289516] Mass Storage Function, version: 2009/09/11
[  362.289542] LUN: removable file: (no medium)
[  362.291329] LUN: removable file: /backingfiles/cam_disk.bin
[  362.291351] Number of LUNs=1
[  362.297153] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  362.297177] g_mass_storage gadget: g_mass_storage ready
[  362.297197] dwc2 20980000.usb: bound driver g_mass_storage
[  362.413205] dwc2 20980000.usb: new device is high-speed
[  362.451296] dwc2 20980000.usb: new address 13
[  362.472466] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[  364.783595] NOHZ: local_softirq_pending 40

skipfire, 2018-10-23:
After a reboot the drive did not reattach, but the modprobe line did make it reappear.

cimryan, 2018-10-23:
That "Archiving..." line means that control transferred to the archvive-teslacam-clips script.

What're the last few lines of the /tmp/archive-teslacam-clips.log file?

skipfire, 2018-10-23:
/tmp/archive-teslacam-clips.log file entries since last boot:

Tue 23 Oct 18:08:30 UTC 2018
Retrying...
Tue 23 Oct 18:08:30 UTC 2018
Mounting /mnt/archive...
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Tue 23 Oct 18:08:30 UTC 2018
Failed to mount /mnt/archive.
Tue 23 Oct 18:08:30 UTC 2018
Sleeping before retry...
Tue 23 Oct 18:08:31 UTC 2018
Retrying...
Tue 23 Oct 18:08:31 UTC 2018
Mounting /mnt/archive...
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Tue 23 Oct 18:08:32 UTC 2018
Failed to mount /mnt/archive.
Tue 23 Oct 18:08:32 UTC 2018
Attempts exhausted.

skipfire, 2018-10-23:
There are no new entries in archiveloop.log.

cimryan, 2018-10-23:
Looks like the Pi isn't able to connect to your archive server.

Can you share your /etc/fstab?

There are a few things that could be going wrong:
Wrong username or password in /root/.teslaCamCredentials
Corruption in /root/.teslaCamCredentials
Archive server misconfigured

You can try making changes, and then running
```
mount /mnt/archive
```

If that works you'll be past this snag.

cimryan, 2018-10-23:
Yes, you'll likely need to edit /root/.teslaCamArchiveCredentials to correct at least the username, if not also the password.

Verifying that the archive can be mounted during setup is a feature that was just added, this morning.

I agree that the failure to mount the archive shouldn't be fatal: https://github.com/cimryan/teslausb/issues/49

skipfire, 2018-10-23:
I'm working now and tried a fresh install as well and it worked.  The only problem I still see is that I got an unrecognized device error and then the it mounted as a USB drive and then error box went away.  This is all using Windows 10.  Now I'm going to try it in the car later and hope it works.

closed 2018-10-23 by cimryan

PhReAkShO, 2019-07-25:
What is the format of the /root/.teslaCamCredentials supposed to be.  I am having this issue with it not being recognized and the modprobe mounts the files.  I do not have the /root/.teslaCamCredentials file.  I do not see anything resembling //192.168.251.16/teslacam /mnt/archive cifs  in my fstab either.