cimryan/teslausb #50: Allow command line parameters in place of prompts

skipfire, 2018-10-23:
Further scripting & automation would be aided by having setup-piForHeadlessConfig.ps1 and setup-teslausb accept command line parameters and prompt for the values only if the command line values are not present.

setup-piForHeadlessConfig.ps1 needs SSID & key
setup-teslausb needs archiveserver, sharename, shareuser, sharepassword, campercent

Perhaps this would help make regression testing easier as well?

skipfire, 2018-10-23:
When I was working on the previous problem, it would have been really nice if I could have just setup a single command line like
setup-piForHeadlessConfig.ps1 e myssid mykey myserver myfolder myuser mypass mysplit
then have it run through the PowerShell which would hopefully even setup a runonce so that the setup-teslausb starts automatically upon booting the Zero and it could reboot at the end.  After running the PowerShell all you do is move the SD card to the Zero and plug it in and let it go and it should finish.

cimryan, 2018-10-23:
Experiments are currently underway to evaluate the feasibility of a single-step process: https://github.com/cimryan/teslausb/issues/39

The official image for the Pi lacks a run-once system, and the work involved in creating and maintaining a new image is significant.

Until someone adds the support you're describing (and please feel encouraged to submit a pull request, yourself), the PowerShell script current supports command-line parameters, directly, and setup-teslausb can be invoked from a script, like this:
```
#!/bin/bash
export archiveserver=foo
. . .
setup-teslausb
```

rtgoodwin, 2018-10-24:
Give this [flashable image](https://www.dropbox.com/s/6f8kxenvtz8pkj9/image_2018-10-24-teslausb_headless-lite.zip?dl=0) a try. Write to card, reinsert it into PC so it mounts `boot` folder. Then create in `boot` a file called `teslausb_setup_variables.conf` and put the vars in it like so (no quotes on any):
```
export archiveserver=your_archive_name_or_ip
export sharename=your_archive_share_name
export shareuser=username
export sharepassword=password
export campercent=100
export SSID=your_ssid
export WIFIPASS=your_pass
```
* Boot it in your Pi, give it a bit, watching for a series of flashes (2, 3, 4, 5, maybe 6) and then a reboot and/or the CAM to become available on your PC/Mac. 
* The Pi should be available at `teslausb.local` over Wifi (if it works) or USB networking (if it doesn't). Takes about 5 minutes for me. 
* I forgot to create the TeslaCam folder, so you'll need to do that before taking to your car. 

LMK if it does/doesn't work for you please, feedback needed! :) Worked great just now; I've only tried it with campercent=100.

rtgoodwin, 2018-10-27:
@skipfire check out my latest comment/image on #39 and see if that style of setup suits you versus prompted arguments. Trying to feel out the user experience and see if just using a `.conf` file works in general better/easier for folks. (It can then be commented and such too, to provide extra guidance.)

skipfire, 2018-12-05:
Closing this out, the OneStepSetup process covers this enhancement.

closed 2018-12-05 by skipfire