cimryan/teslausb #54: Need to Update Message to Users About Setting CIFS Version

ken830, 2018-10-24:
Tried to mount a share folder on my Synology NAS. Got the "mount error(95): Operation not supported" error when running `setup-teslausb`. Message to the user suggest this: "

> Try specifying a lower number for the CIFS version like this: export cifs_version=2

This actually doesn't work. I suggest changing this to:

> Try specifying a lower number for the CIFS version like this: export cifs_version=2.0

This works.

I'm a h/w guy and this is my first time really on github... If it's alright with you guys, I would like to try my hand at contributing with this minor change. Would I just open a pull request? Sorry for being a total newb.

ken830, 2018-10-24:
@rtgoodwin 👍 That was super-helpful for me. Thank you. I think I've got the PR done.

I'll stick with web editing for now. I tried git from command line about 10 years ago. Boy, I'm old...

ken830, 2018-10-25:
Okay, I thought I did the right thing, but now I'm not so sure... I don't see my pull request here on cimryan/teslausb. I only see a pull request on ken830/teslausb.

cimryan, 2018-10-25:
Merged!

closed 2018-10-25 by cimryan