cimryan/teslausb #63: Automatically determine CIFS version

cimryan, 2018-10-25:
When the archive can't be mounted we can't currently determine whether the problem is the CIFS version or the username/password.

verify-archive-configuration.sh could be replaced with another script, say, "get-archive-configuration" which would return the mount options, or at least the CIFS version.

This script could iterate through the four different CIFS versions that mount.cifs supports, and fail if none of them work instructing the user to check the username/password.

tonychi8, 2018-11-04:
I ran into this error.  Using a folder share on Mac.  
First error (permission denied from the script) because I only turned on SMB.  
Then I turned on AFP with user password added to it.  That gave me (invalid argument ) error.

[image]
 
[image]

[image]

cimryan, 2018-11-04:
I've just added more logging to this step. If you try again you'll see the command-line that the system used to try to mount the share. That may provide some insight.

You might also try running
```
dmesg | tail -n 30
```
after the failure to see if that provides any more useful data.

When this failure occurs does the content of /tmp/teslaCamArchiveCredentials look correct? Are there any special characters in the username or password? Any backslashes or quotes?

Did you try specifying different `vers` values?

tonychi8, 2018-11-04:
hum.  it seems the script does not like the double quotes around the server name (I used IP address with double quotes around it, it worked yesterday but not anymore).  Changed to IP address without double quotes and it got pass the step. 
Now it is failing the same place.  This time, the command line has the share name with double quote (just as how it is entered - I used the "SailfishCam" as the actual share folder name to keep it simple.)

The /tmp/teslaCamArchiveCredentials has double quotes around the user name and password.
With double quotes, it fails with permission denied.
Without the double quotes (I manually edit the file), it fails with invalid argument.

error:
[image]

dmesg: 
[image]

error after removing double quotes for user name and password:
[image]

cimryan, 2018-11-04:
Everything here looks like a bad username/password combination.

You might try connecting using those credentials from another computer, or your phone.

tonychi8, 2018-11-04:
I had to get a windows computer to figure this one out.  You are on the right direction.  It seems the Mac Share is seen as the same name as the user (not the name of the folder).  So, from windows computer, it sees the share as tesla (user name) and not SailfishCam (folder name).  It works once that is fixed (I also removed all the double quotes for the four exports statements).
Different question: does this version do the web site?
Thanks for all the help.

alanhoyle, 2018-11-07:
In a related bug, if CIFS fails, you can't rerun the previous command and have it resume.  It tells you that there isn't enough space on the card and quits instantly.  Rather than try to debug, I reflashed Raspbian.