cimryan/teslausb #67: FIles are removed rather than copied

skipfire, 2018-10-26:
I have been finding that sometimes the files deposited on my server are only 2K in size, and since they are purged from the PI there is no recovery.  Given that these may contain important video the default should be to preserve it on the Pi and just copy to the file server, then if there are any problems the originals are still available.

cimryan, 2018-10-26:
The problem here is that the Tesla cuts the power to the drive before completes writing to the drive. So, the data beyond that 2K is just lost, and the file is actually just 2K in size.

The Tesla won't start writing to a file that was previously written to, that time has already passed.

So, copying the file instead of moving it would just copy a corrupt file instead of moving one.

There are a couple of things we could try to reduce the corruption:
* Apply the "sync" option to the /backingfiles mountpoint in /etc/fstab
* Periodically issue "sync" commands to the filesystem.

The second options seems like a hack. The first option requires using ext3 for the backingfiles partition, I think, since it's my understanding that ext4 doesn't support the sync option.

skipfire, 2018-10-26:
Just not deleting files is my request, not the 2K aspect.  If there is a minor accident (no ambulance and cars can still drive), I want the maximum chance of not losing files, so a copy instead of a move would be preferred.

cimryan, 2018-10-26:
I suspect that, whether or not the file is deleted, it will still just be 2K in size.

We know that files aren't deleted until they're fully transferred. You can confirm this by writing a large file into the TeslaCam directory and naming it, say, saved-1.blob. You can watch your network share to see it start to copy. While it's copying, pull the power from the Pi. Then plug the Pi in, again. It'll start copying the file, again, and copy the whole thing to the share, if you let it, overwriting the bad file with the whole good file. If you do the same thing, deleting the partial file from the share before you plug the Pi back in, you'll see the whole file get copied to the share. So, none of the data in the file was removed from the Pi until the whole file was on the share.

The advantage of this move operation over a copy operation is that the system automatically reclaims space. If we copy instead of move, we have to implement some process to free up space.

It looks like we're going to be adding a partition which is read/write on the SD card. We can move the logs to that partition and we can record the file sizes of the files before they're moved. If we find that the file sizes are different from what we're seeing transferred then we'll know that we're not getting the data. If the file sizes are the same, though, we'll know that copying instead of moving won't provide a benefit.

In the mean time, if you'd like to try running with copy instead of move, you can edit /root/bin/archive-clips.sh and replace the mv command with a cp command.

cimryan, 2018-10-27:
If you're going to copy the files instead of move them you might also want to implement some way of avoiding copying files that have already been completely copied.

skipfire, 2018-10-27:
My thinking on that was to copy them, then rename from saved* to copied* or stored*

cimryan, 2018-11-04:
The archiveloop.log file now persists between power cycles at /mutable/archiveloop.log