cimryan/teslausb #68: Feature: combine files before copy/move

rtgoodwin, 2018-10-26:
I'm still thinking this one through, but it's quite annoying that files are 1min / 30MB in size and having to deal with them after they get copied off. It'd be nice to have an option to combine files within a range (like everything for a given day stamp) before copy.

FFMPEG can concatenate, given a list of files, so it would be something like "for every file with name like %date% and file size == 30.1MB" written to a file and FFMPEG cats them. 

This may never make sense unless you're doing the climate control trick, or a passthrough battery trick. You need to ensure file integrity and so on, obviously. 

Putting out there to solicit feedback either way.

cimryan, 2018-10-27:
I think this idea could be made to work.

I like that with smaller clips I can get a few transferred every time I get in or get out of my car. I think I would rather do my video editing on my computer.

I'm a little concerned about the complexity this would add to the project.

Those are my opinions, but as the maintainer of the repo I'll keep an open mind. If this gets interest and the implementation doesn't seem like it'll add a bunch of maintenance cost or risk of regression I won't stand in the way.

rtgoodwin, 2018-10-27:
Yeah it's definitely not the ideal venue for it. And I would think Tesla would fix the behavior once they've got some time under their belts with the cam software/routines.

It'd probably be just as good to have something sit on the archive side and monitor and stitch. I've got some ffmpeg time under my belt so maybe I'll tinker around there. The challenge is more the logic for grouping the filenames together and such, which I'm sure will come to me at some point. 

Maybe leave this one open a couple days and see if others have ideas or have tackled the problem in any way? Then we can kill it and we'll see if it really _is_ bothering me enough to fix it one way or another. 😄

skipfire, 2018-10-27:
I would not suggest combining the files, while it may be nice during playback, if you are using it for evidence then someone could complain that the files have been edited and it increases the likelihood of ending up with corrupted files.  Looking at my valid files there seems to be about a 800kB spread of file sizes which is more than 1 second of video and if the power down happened at 59 seconds you could have a final file that is big enough, but still not valid since it did not finish.  Too bad Tesla didn't use simpler mpeg or ts files, they would be bigger but they tend to be usable even when prematurely terminated.

rtgoodwin, 2018-10-27:
Yeah I don't love it. I'm going to go ahead and close this and find a better solution for the actual desired outcome. :)

rtgoodwin, 2018-10-27:
And agree, the way files are written is a mess. (Or, to be more charitable to what is a "free" feature, it is "unoptimized".)

closed 2018-10-27 by rtgoodwin