cimryan/teslausb #69: Multiple USB media drives appear in vehicle

ken830, 2018-10-27:
So, the dual drives was working great for me for about a day. Now, I see a bunch of USB drives in the media app. The first time I noticed this, I tried to click one of the "drives" just off the top of the screen and I could see my music, but now, even that doesn't work. I suspect the car keeps seeing progressively more and more empty drives and the one with music is now pushed way off the edge of the screen. Or I could be totally wrong.

[(https://imgur.com/BbaiWyu)]

cimryan, 2018-10-27:
I've switched the ordering of the files in g_mass_storage.conf. This causes the music drive to appear in the view in the Tesla. Just pick the device that has songs listed on it.

I think we're going to have to live with multiple devices appearing in the list in the Tesla. The only way I've found to eliminate having two devices for each backing file is to use removable=0 in g_mass_storage.conf, but that seems to cause Windows to not see the USB drives.

cimryan, 2018-10-27:
If someone can figure out how to show only one device in the Tesla, and still be able to add/remove files when the drive is plugged into a PC, and still have the automatic transfer work, please submit a PR.

ken830, 2018-10-28:
I saw the reversal of the drives in `g_mass_storage.conf`. Is there any way I can switch the order myself by editing the file? The filesystem is read-only when I SSH in and try to edit it.

cimryan, 2018-10-28:
Instructions for making changes are at the bottom of the main readme.

ken830, 2018-10-28:
Thanks. I don't have that script in there:

`root@teslausb:/# /root/bin/remountfs_rw
-bash: /root/bin/remountfs_rw: No such file or directory`

but I was able to manually type the commands to remount after looking at the script in the code section of the repository.