cimryan/teslausb #71: Add website for configuration

skipfire, 2018-10-27:
I plan on working this one, just wanted to open it for feedback.

Utilizing apache2 & php, have a simple web UI running on the Pi that can provide some statistics (such as space used/free), mount status, and then enable control of new settings to drive minor variations in script behavior such as copying files instead of moving them.  Potentially also add the ability to trigger the copy process manually if it did not fire for some reason.

rtgoodwin, 2018-10-28:
I was looking at Filebrowser for some nice basic file access, downloading. and viewing features. It's cool because it's a single binary, and might be extensible (and I've been meaning to learn Go). Didn't dig into how extensible it was. Before that I was considering an nginx or similar tiny server, default install nice and lightweight, to manage or change some configuration aspects. All that to say I might be able to kick in on this. It also supports the base image strategy; if you can flash an image that comes up in USB networking mode with wev server running, you're set for lots of neat scenarios. LMK if I can help in any way.

fat-tire, 2018-10-28:
I was gonna say, isn't apache a bit heavy for this?

skipfire, 2018-10-28:
@fat-tire the apache2 install is really small, though it may have quite a bit of capability that isn't needed.  Though I don't think that is necessarily bad as there may be other thinks people may want to do with a Pi in their Tesla.  Combine that with how little I know about Linux (I'm a Windows developer) and that I do know PHP and it fit.

Right now the biggest problem is that the TeslaUsb has /var/log as tempfs so the directory gets wiped out every boot and then apache fails to load.  I did remove that line from the config so now it sticks around so I just need to start figuring out the hard stuff, like issuing the correct commands with the correct rights and not making a huge security mess out of it.

rtanaka, 2018-10-28:
@skipfire - ran into a similar issue with getting things to work due to not having a writeable partition. @cimryan and I have a branch where a writable partition will be introduced (/mutable) hopefully that wil get merged into the main soon so we can all start using it. 

i'll leave it to you guys to figure out what front end to use. +1 on making it as light weight as possible though.

cimryan, 2018-10-29:
Similar to the way that the different archiving technologies have now been factored out of the main readme.md, I think there's an opportunity to inject between steps 3 and 4 in "Set up the USB storage functionality" a step which lists various enhancements with links to ancillary instructions.

ppamidimarri, 2019-02-25:
> I was looking at Filebrowser for some nice basic file access, downloading. and viewing features. It's cool because it's a single binary, and might be extensible (and I've been meaning to learn Go). [...]

I have built a web browser using Responsive File Manager on an nginx server. I don't know how to post the code here. I have the code and am willing to share it.