cimryan/teslausb #77: "UPS" support?

abl, 2018-10-30:
I was thinking about the problem of powering the Pi when the car turns off - long enough to upload.

The best solution I could find was [this HAT by Alchemy Power](https://www.amazon.com/Alchemy-Power-Inc-Pi-Zero-UpTime-devices/dp/B073C3NJ2T). With an 800mAh battery in place, given ~250mA max draw on the Pi, that's still plenty of power even assuming fairly heavy losses. I'm planning to ignore the low battery GPIO - if you shut down on low battery, there's no signal to restart unless you press a button. I'll know more tomorrow when it arrives; for now, it seems to be pretty much the same thing as using a splitter and connecting a USB battery.

Assuming shutdown isn't an option, is there anything useful to do when we know the car's off? (Corollary problem: the car might come back on before we lose power, so there's no guarantee of a clean boot.)

I'm also planning to build a watchdog circuit for this problem unless Tesla ships automatic camera upload before I have time to think about the problem - I'll publish all of my work/research in a fork under the same license, of course!

cimryan, 2018-10-30:
When the power is cut to the USB ports the Model 3 definitely can't be writing to the drive, so it would seem to be an ideal time to upload clips to the archive.

I considered pre-ordering one of these: https://www.crowdsupply.com/silicognition/lifepo4wered-pi-plus

I decided I'd be too worried about the battery exploding.

skipfire, 2018-10-30:
I'm working on seeing if I can get a HAT working that would provide 30 to 60 seconds of power and the knowledge that the outside power had been cut, hopefully giving enough time for it to finish writing a file and then do a controlled halt, no sure if I can make it work to a usable time and size since I'm wanting to use caps instead of a battery.  The caps would be nice for a very short recharge, but the size may be prohibitive.

lolento, 2018-10-30:
I have a Xiaomi Power bank that works as a passthrough when being charge.  So I basically charge the battery bank via USB and then use the battery bank to power the PI.

Kind of already tried this out...but I loaded an early version of this script (from the very beginning)

My observations:
- Somehow when the pi is powered by battery, clips are still recording to the drive when my car is off.  
- With the early version of this script, because the pi is not going thru a power cycle, I get the dreaded "X" corruption logo on the Tesla

It didn't go well for me.  But I haven't gave it another shot with all the script upgrades...

skipfire, 2018-10-30:
A rechargeable battery solution will not work well for this without some extra piece to shut it down.  You will drain that battery completely, then recharge it a little on your next drive, then discharge it completely again which will not take long since it will not get the longer duration recharge.

rtgoodwin, 2018-10-30:
There are some good posts about this: https://learn.adafruit.com/adafruit-powerboost-1000c-load-share-usb-charge-boost/overview

Basically this + a LiIon or LiPo which they also sell, and possibly very cheap voltage controller they sell (although the forum suggests the PowerBoost will auto detect low voltage.  So it will power + passthrough charge, switch to battery, back and forth as needed. I haven't ordered it yet but might.

DeweyVT03, 2018-10-31:
I went with this as a solution, but modified what I connected to the GPIO and did not use a button switch.
https://github.com/NeonHorizon/lipopi

I have it go high enabled when power is supplied via the USB (car turned on). This then boots up the Pi and it begins monitoring for the low power state to plan for a graceful shutdown.

Achieved this on the PowerBoost board by connecting the USB pin to the EN pin. Also connected to the EN pin is the Pi GPIO through a 10K resistor (as shown in lipopi above) and 100K resistor to GND (as shown in lipopi above) to flip it to off when the other two no longer give a high signal. The LB on the PowerBoost board is connected to the Pi GPIO as shown in the demo above.

Only thing I need to do to improve it is to change the cron script to trigger a graceful shutdown on two conditions: either low battery or file transfers all done.

cimryan, 2018-10-31:
> I went with this as a solution, but modified what I connected to the GPIO and did not use a button switch.
> https://github.com/NeonHorizon/lipopi

@DeweyVT03 That sounds like a nice fit. If you have a page with more details I'd be happy to link to it from the main readme.

cimryan, 2018-11-02:
I'm going to close this "issue", for now. If someone gets a UPS working please submit a PR.

closed 2018-11-02 by cimryan

rtanaka, 2018-11-06:
came across this solution -- if i go down this route, this is likely what i'll try: https://www.sparkfun.com/products/14803

lolento, 2018-11-06:
I actually ended up creating an geo-fenced HVAC automation with Homeassistant's Tesla component.  Turn's on the climate for 20min each night when Tesla is at home.  No extra hardware needed.

cimryan, 2018-11-06:
I've added these ideas to a wiki: https://github.com/cimryan/teslausb/wiki