cimryan/teslausb #80: OneStep: Interactive prompt during partitioning

rtgoodwin, 2018-10-30:
Thanks for the merge! 

Just tested and looking good except for a minor change I'll be pushing to the image to default to your master branch. If using the conf sample file you'd probably be fine, but in mine I had fully commented out the repo/branch which made the initial script download fail. 

But this prompt happened during a manual run where partitioning which will stop the interactive setup I think: 
```
Warning: You requested a partition from 63757999616B to 63864569344B (sectors 124527343..124735487).
The closest location we can manage is 63758663680B to 63864569344B (sectors 124528640..124735487).
Is this still acceptable to you?
Yes/No? 
```
First time I've seen this, I'm guessing it's not going to be hit for everyone, or maybe a recent change?

I added the `-s` flag  ("script"/non-interactive) to `parted` in a new branch `more_logging` but `parted` doesn't consider this a passable error.  
```
Starting to create backing files partition...
Checking existing partitions...
Modifying partition table for backing files partition...
Modifying partition table for mutable (writable) partition for script usage...
Error: You requested a partition from 63757999616B to 63864569344B (sectors 124527343..124735487).
The closest location we can manage is 63758663680B to 63864569344B (sectors 124528640..124735487).
```

That's from this morning's codebase... do we need a slightly more conservative set of partition boundaries, or maybe it's a calculation bug?

cimryan, 2018-10-30:
Internet Lore suggested that decorating the partition boundaries with M would allow parted to round the address to an aligned value.

I wonder what would happen if we used BACKINGFILES_PARTITION_END_SPEC as the start address for the mutable partition.

rtgoodwin, 2018-10-30:
New image release (9b2) up: https://github.com/rtgoodwin/teslausb/releases/

rtgoodwin, 2018-10-30:
> I wonder what would happen if we used BACKINGFILES_PARTITION_END_SPEC as the start address for the mutable partition.

I tried this and it seems to work, at least doesn't complain. Haven't inspected too much yet:
`MUTABLE_PARTITION_START_SPEC="$BACKINGFILES_PARTITION_END_SPEC"`

rtgoodwin, 2018-10-30:
Results on a 64GB card:
```
pi@teslausb:~ $ df -h
Filesystem                    Size  Used Avail Use% Mounted on
/dev/root                     1.6G  1.1G  485M  69% /
devtmpfs                      213M     0  213M   0% /dev
tmpfs                         217M     0  217M   0% /dev/shm
tmpfs                         217M  5.8M  211M   3% /run
tmpfs                         5.0M  4.0K  5.0M   1% /run/lock
tmpfs                         217M     0  217M   0% /sys/fs/cgroup
tmpfs                         217M   12K  217M   1% /var/log
tmpfs                         217M   12K  217M   1% /tmp
tmpfs                         217M     0  217M   0% /var/tmp
/dev/mmcblk0p1                 44M   22M   22M  51% /boot
/dev/mmcblk0p3                 57G   54G     0 100% /backingfiles
/dev/mmcblk0p4                 94M  1.6M   86M   2% /mutable
//MY/SECRETSHARE  7.3T  5.4T  2.0T  74% /mnt/archive
tmpfs                          44M     0   44M   0% /run/user/1000
```

cimryan, 2018-10-30:
Looks good. Want to create a PR for a bugfix?

rtgoodwin, 2018-10-30:
😎 #82

cimryan, 2018-10-30:
Merged.

closed 2018-10-30 by cimryan