cimryan/teslausb #86: Direct access to saved clips on the go

cimryan, 2018-10-31:
Imagine that the car is in an accident outside the range of a cell tower. You'd like to get the video onto your phone so you can show the local police what happened. You don't have a laptop in the car.

It should be possible to configure the Pi so that it acts as both a WiFi client and AP at the same time: https://www.raspberrypi.org/forums/viewtopic.php?t=138730

With just that change the scenario would be achievable with a terminal client and an SFTP client running on the phone.

A further enhancement would be to add a simple web server on the Pi which allow one to trigger
1. (virtual) ejection of the USB drive from the car
2. mounting the backing file

A yet further enhancement would be to list the saved clips through the website for direct download to the phone.

rtanaka, 2018-10-31:
i've actually used a Pi for something similar to what you're discussing. while it is possible for the Pi to act as either a client or an AP, it cannot do both at the same time. so doing something like you describe would be a little tricky since the syncing requires that it be in client mode waiting for a AP to connect to. unless someone has a clever way of switching between the two modes, short of adding a second WiFi dongle (which i am not a fan of) i'm not sure we can do something here.

that said, i've actually been thinking about this quite a bit and have a couple options i'd like to discuss

1. add support for having a non-syncing wifi connection for phones with mobile hotspot. i don't think it makes sense to sync over mobile but we can at least use this for emergency purposes like the scenario you described. in this mode we'd be able to go to a webpage where we can sync selective files just as you described.

1. leverage bluetooth somehow. i *feel* like this is the most elegant way... unfortunately, i'm not as well versed on leveraging this. i plan on doing some research here but i *feel* like we can use bluetooth.

looking forward to other thoughts...

rtgoodwin, 2018-10-31:
I was going to start dropping this single binary webserver into the images as an option for those that wanted to use or enable it: https://filebrowser.github.io/  It has basic file download capabilities that I THINK work on mobile browsers but I'd have to test again to be sure. 

You can use something like [networkd-dispatcher](https://gitlab.com/craftyguy/networkd-dispatcher) to track the state of your network interfaces and run specific scripts based on the state, ESSID, etc. (I have to double check that systemd-networkd is working on the base lite image.) This could help start up the binaries based on the SSID you're on, etc. This may be helpful for #17  VPN thread that I haven't gone back and read TBH. :) @cimryan is it possibly even a replacement for the archiveloop `ping` methodology? This way it's sleeping basically until the network state changes...probably not a ton of overhead, but the less things you have to control the better. 

I haven't explored BT at all but data rate over BT I think is pretty abysmal...then again it is an emergency as you mentioned.

rtgoodwin, 2018-11-04:
Over on the Slack I discussed with @rtanaka a method about pairing your phone to the Pi over BT, with a BT PAN set up. You could then load a page from your phone browser to trigger the Pi to switch into AP mode (and unmount the mass storage) to load a web page showing the clips. It's not super unwieldy, and I think it could work. Open to ideas there. 

Loading even a moderately complex page over BT PAN seemed to be very slow; I'll have to do some more testing to see just how slow it is.

ifnull, 2019-08-03:
I'm looking into adding a [switch to the GPIO](https://github.com/masteruan/Raspbery_switch) to turn on a [hotspot](http://www.raspberryconnect.com/network/item/330-raspberry-pi-auto-wifi-hotspot-switch-internet) and a [captive portal](https://github.com/nodogsplash/nodogsplash).