cimryan/teslausb #87: The setup script assumes the file share is pingable

Cisien, 2018-11-01:
I am attempting to set this up to upload files to an azure storage file share. I can manually mount the share on the pi, however, it fails when using the same values in the setup script.

What is the reason for using the IP address instead of the name supplied?

I'm planning on modifying this in my repo to skip this and use the name instead of IP. I can PR that change if desired.

cimryan, 2018-11-01:
The principal reason is because that's how it the script was written by the original author and we haven't found a need to change it.

Looking around on the internet, I could speculate that perhaps the reason it was written that way was in case the mount happened before name resolution happened. But I've read that there's a "_netfs_ option that could also address this.

It sounds like you're saying that your archive server
1. Doesn't have a stable IP address
2. Isn't pingable

Is that right?

Cisien, 2018-11-01:
I'm not sure how stable the IP address is, since this is fully Azure hosted, but they definitely block icmp ping on that address.

Cisien, 2018-11-01:
it looks like the 'hping3' package can be used `hping3 -c 1 -S -p 445 $hostname` instead in order to test for connectivity to a cifs share.

cimryan, 2018-11-01:
Does hping3 work for all CIFS shares, on Windows, MacOS, Samba, others? If so, it seems reasonable to replace the ping test with that test for CIFS archives.

The alternative would be to add Azure File Share as another hosting technology. The current organization of the files actually makes that option easier, but it would be nice to consolidate all the CIFS options.

Cisien, 2018-11-01:
I have only tested with azure's file shares, but I believe it should work. It works by using a TCP SYN to test if the port is open and listening.

cimryan, 2018-11-02:
If you'll submit a PR I'll be happy to merge it into a branch in this repo to increase the chance that someone will try out the change with a Windows share.

Cisien, 2018-11-04:
https://github.com/cimryan/teslausb/pull/92

I've done some very basic testing locally with my windows 10 pc, and a freenas samba server I have running, and hping3 worked well.

cimryan, 2018-11-04:
Merged!

closed 2018-11-04 by cimryan