cimryan/teslausb #100: Is there a simple way to search and delete the corrupt video files that has been archived to a share?

lolento, 2018-11-11:
I guess Tesla has a habit of recording corrupted video.....at least 1 file per day sometimes more.

Is there an easy to search and delete them all?

lolento, 2018-11-13:
Figured out a solution unless someone has a better one.

Get mediainfo CLI (unpack to C;\Util\MediaInfo) and use the following windows batch script:

```
@echo off
set "exe=C:\Util\MediaInfo\MediaInfo.exe"
for /r %%a in (*.mp4) do (
"%exe%" -f "%%a" |find "Height" | findstr /r "[960]" >nul && xcopy "%%a" /y X:\Surveillance\Model_3
)
del /f /q .\*.mp4
```

This copies only the video files that MediaInfo has determined to have a frame height to the X:\Surveillance\Model_3 folder from the current folder and then deletes all mp4 files from the current folder.

rtanaka, 2018-11-14:
these are corrupted before they even get to the server, right? seems like we should just check before we try and upload.