cimryan/teslausb #102: 64gb drive shows insufficient space

ouhakheme, 2018-11-23:
Formatted the card with Etcher
Setup the pi using the Windows Powershell Instructions
Logged in via wifi and setup rclone
Ran setup-teslausb and received this message:

Downloaded /tmp/verify-configuration.sh ...
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small.

parted shows the following:

/dev/mmcblk0:64021856256B:sd/mmc:512:512:msdos:SD GC2QT:;
1:32256B:4194303B:4162048B:free;
1:4194304B:50119679B:45925376B:fat32::lba;
1:50119680B:50331647B:211968B:free;
2:50331648B:1866465279B:1816133632B:ext4::;
3:1866465280B:63915950079B:62049484800B:ext4::;
4:63915950080B:64021856255B:105906176B:ext4::;

ouhakheme, 2018-11-24:
Problem with the SD Card, my fault.

closed 2018-11-24 by ouhakheme

lborloz, 2018-12-30:
Having a similar problem. What was the solution?

luzer, 2019-07-14:
i also have this problem...

ouhakheme, 2019-07-14:
My solution was just to reformat the card, no errors after that.