cimryan/teslausb #106: One Step Setup stalled

skipfire, 2018-12-05:
I am using the OneStepSetup and it seems to have stalled on "Updating package index files...".  I have TESLAUSB_SETUP_STARTED and WIFI_ENABLED files.  The setup log is below.  The date/time has updated since, but it has been 40+ minutes since I first saw updating package index files.  I am able to ping my share server as well as yahoo, so network config is good.  Is there anything else I can look at or some way to spur it beyond updating package index files?

Tue Oct 30 18:39:39 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Tue Oct 30 18:39:39 GMT 2018 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Tue Oct 30 18:39:39 GMT 2018 : Rebooting...
Tue Oct 30 18:40:05 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Tue Oct 30 18:40:05 GMT 2018 : Grabbing main setup file.
Tue Oct 30 18:40:06 GMT 2018 : Starting setup.
Tue Oct 30 18:40:06 GMT 2018 : Updating package index files...

skipfire, 2018-12-05:
I did try rebooting it a bit later and it went right to the Updating package index files, even waiting overnight hasn't seen it move.

pi@teslausb:/boot $ cat teslausb-headless-setup.log
Tue Oct 30 18:39:39 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Tue Oct 30 18:39:39 GMT 2018 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.

Tue Oct 30 18:39:39 GMT 2018 : Rebooting...
Tue Oct 30 18:40:05 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Tue Oct 30 18:40:05 GMT 2018 : Grabbing main setup file.
Tue Oct 30 18:40:06 GMT 2018 : Starting setup.
Tue Oct 30 18:40:06 GMT 2018 : Updating package index files...
Wed Dec  5 00:41:56 GMT 2018 : Detecting whether to update wpa_supplicant.conf
Wed Dec  5 00:41:56 GMT 2018 : Starting setup.
Wed Dec  5 00:41:57 GMT 2018 : Updating package index files...
pi@teslausb:/boot $ date
Wed Dec  5 12:21:16 GMT 2018

cimryan, 2019-01-14:
If your WiFi password contains special characters and escape escape any single quote characters. I've updated the instructions at https://github.com/cimryan/teslausb/blob/master/doc/OneStepSetup.md to include these instructions.

If this doesn't address the problem, please reopen this issue.

closed 2019-01-14 by cimryan