cimryan/teslausb #109: A couple of bugs/enhancements

raheelsoz, 2018-12-31:
First off, Thanks for creating this repository, its very helpful.

I'm new to linux (develop mostly on windows (.Net)) so it was a very interesting experience learning how to work with a distribution and PI in general.

I ran into a few snags while setting up and it took me a decent amount of time to figure out what was going wrong. I'll list each below, it may help others running into something similar.

1) Hidden SSID, will not connect to Wifi:
This took me a while to crack. I burnt tens of raspbian and OneStepSetup images to identify why it would work in 1 scenario and not the other.

It all boiled down to the scan_ssid=1 variable not being present in the network object for wpa_supplicant.conf.

For my hidden (non broadcasting SSID) it only works when this field is present. I believe this should be mentioned in the tutorial and coded into the OneStepSetup. Will help people like me make sense of the failures.

2) Unescaped '!' in the variable values can cause the script to break.
The second issue I ran into once I figured out the Wifi was, usb drives would not show up. The cam and music files were not mounted and the archive settings were missing in /etc/fstab (as well as the creds file ) using the OneStepSetup.

When I tried the manual setup and executed the export command for the password with '!', it returned an error which made me realize that '!' was being considered an operator.

I would suggest this be added to the documentation as well so people new to linux know how to properly used escape character and avoid causing the setup to fail.

3) ArchiveLoop does not seem to be running:
Once I fixed both of these issues, I placed a test file named "Saved-test.mp4" with some garbage data to see if the file will transfer to the archive share.

I was unable to get the file to move to the share. Not sure what I'm missing, I tried to manually execute /root/bin/archiveloop but no luck.

Also, there are no logs in the /tmp folder for archiveloop. Although, going through the code and doing a file search I was able to find that the archiveloop.log file is being written to /mutable/archiveloop.log

I'd appreciate some feedback and direction on how to get this to work. I'll try plugging it into the car and see what happens.

Thanks again for all your work.

wwwb0n3zcom, 2019-01-12:
I have a '!' in part of my user password and within the wifi password. So this is within the file: teslausb_setup_variables.conf

Is this going to be a problem? So far I have not been successful at completing this using a 128GB or 200GB MicroSD card.

If there is a fix - can someone please post details what is needed or show an example?

Thank you

cimryan, 2019-01-15:
You'll need to escape special characters in your password. Prepend a backslash to each symbol.