cimryan/teslausb #110: Fix for missing .rclone.conf

raheelsoz, 2019-01-01:
I was able to successfully upload Cam clips to DropBox using RClone. The only issue I ran into was that /root/bin/archive-clips.sh was pointing RClone's config to /root/.config/rclone/.rclone.conf which did not exist.

Changing this to the actual location of the config @ /root/.rclone.conf (after making the mount write-able (/root/bin/remountfs_rw)) allowed the archival process to start successfully.

rtanaka, 2019-03-07:
@raheelsoz - fyi -- rclone periodically refreshes the token so it needs to be in the /mutable directory or it will fail at some point. i recommend you keep it the way it was set up.

DeweyVT03, 2019-03-29:
rclone doesn't appear to be putting the config file where this code source is expecting it?
Looks like the code thinks it will be at /root/.config/rclone/.rclone.conf
But rclone config writes it to /root/.rclone.conf

Does this need to be moved to /mutable and then create a link using ln -s ?

overtone1000, 2019-06-15:
I encountered this problem with [v1.2 of the marcone fork](https://github.com/marcone/teslausb/releases/tag/v1.2a). The following fix worked for me.
`/root/bin/remountfs_rw
export RCLONE_CONFIG="/root/.config/rclone/rclone.conf"
rclone config`
After which I obviously had to repeat rclone configuration, but you could just as easily move your rclone configuration file to the appropriate directory.

It would be helpful to add `export RCLONE_CONFIG="/root/.config/rclone/rclone.conf"` to the [instructions for RClone setup](https://github.com/cimryan/teslausb/blob/master/doc/SetupRClone.md). It should be performed prior to executing `rclone config` so that the configuration file is created in the directory that the teslaUSB scripts expect.

overtone1000, 2019-06-16:
@rtanaka, I see what you mean about the need for the config file to be in the mutable directory. The archiveloop log contained warnings that it wasn't able to modify the config file. I've done as you suggested in issue #123, which is to move the rclone.conf file to the mutable directory and put a symlink in the `/root/.config/rclone` directory. I'll keep an eye on the log to make sure these warnings no longer occur.

My pi instance doesn't have the `/root/bin/configure-archive.sh` script you referenced in issue #123, so no surprise those changes weren't made during configuration. I wonder if this is a problem with the [marcone fork](https://github.com/marcone/teslausb/releases/tag/v1.2a). marcone did explicitly state that fork won't support archiving methods other than CIFS.

rochec, 2019-07-23:
@overtone1000 do these steps need to happen prior to running TeslaUSB setup? I'm getting this error  Failed to save config after 10 tries: failed to create config directory: mkdir /root/.config/rclone/: file exists

I've already run setup though and ran into the missing config again.

overtone1000, 2019-07-23:
@rochec I made the above rclone configuration changes without repeating TeslaUSB setup afterward. Just to be doubly transparent, I used the marcone fork of this repository.