cimryan/teslausb #113: missing file: /root/bin/remountfs_rw

wwwb0n3zcom, 2019-01-13:
https://github.com/cimryan/teslausb#optional-making-changes-to-the-system-after-setup

File seems to be missing according to the instructions.

rtanaka, 2019-03-08:
i see it in my installation. and i can see that [`the script file`](https://github.com/cimryan/teslausb/blob/master/run/remountfs_rw) is copied as part of the set up process in [configure.sh](https://github.com/cimryan/teslausb/blob/master/setup/pi/configure.sh)

iceraven101, 2019-03-29:
I've yet to get remountfs_rw to actually be copied anywhere after running several install attempts.

jfrux, 2019-05-10:
I'm also not able to get this file... any tips?

laudbrian, 2019-06-02:
you guys should all be using this fork as it's more up to date:
https://github.com/marcone/teslausb,
I have been using this flawlessly the last few months.