cimryan/teslausb #117: error while running ./setup-teslausb

milosGR, 2019-01-28:
Starting to create backing files partition...
Checking existing partitions...
Modifying partition table for backing files partition...
Error: The location 8017412096 is outside of the device /dev/mmcblk0.
root@raspberrypi:~#

milosGR, 2019-01-28:
some additional info; 

root@raspberrypi:~# sudo parted -l
Model: SD LEXAR (sd/mmc)
Disk /dev/mmcblk0: 8017MB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  50.2MB  46.0MB  primary  fat32        lba
 2      50.3MB  8017MB  7967MB  primary  ext4

root@raspberrypi:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7628232 1427100   5858268  20% /
devtmpfs          217616       0    217616   0% /dev
tmpfs             221936       0    221936   0% /dev/shm
tmpfs             221936    3096    218840   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             221936       0    221936   0% /sys/fs/cgroup
/dev/mmcblk0p1     44220   22541     21679  51% /boot
tmpfs              44384       0     44384   0% /run/user/1000

milosGR, 2019-01-29:
so I re-did the whole thing using headless setup and it successfully installed, however when I plug the pi into my tesla I don't see the icon showing that it's recording ... anything I'm missing ?

milosGR, 2019-02-01:
Figured out the issue, I was plugging the microSD into a USB thumb drive that was screwing up the partitions I guess.  Used the adapter that came with the microSD and now everything installed OK.

closed 2019-02-01 by milosGR

Rykemapo, 2019-02-08:
@milosGR I am having the same issue. How do you do headless setup?

siquenique, 2019-04-04:
I am having the same issue, did headless setup solves it? How can we trigger it?