cimryan/teslausb #119: Model 3 doesn't recognize the drive after 2019.5.x

ppamidimarri, 2019-02-25:
I got a grey X on my Model 3 display after the 2019.5.3 update. I rebuilt my drive again (and now I am on 2019.5.4) but same error. 

I am setting up my drive using an older version of setup-teslausb after setting the environment variables for rclone (and campercent=60). I am setting up my Pi Zero W first to have an nginx webserver and then running setup-teslausb to create the partitions and make the / filesystem read-only. This method worked reliably on 2018.x.x versions of the car software.

The drive works properly when my Pi Zero W is plugged into my Windows 10 laptop. But it won't get recognized by the car. Is this a known issue, and is there a different way to create the partitions now?

lolento, 2019-02-26:
Same here!

Somebody please take a look.  I was very happy with this setup until the recent Tesla Sentry Mode update.

acostein, 2019-03-03:
Same here... Model 3 on 2019.5.4 4c3c414. Tried a few times with 3 different SD cards (latest one a SanDisk Ultra 400 GB with class 10 / A1 / U1 rating). On first plugin after setting up the Pi, it looks like it might work for a second, then grey X. Next plugin, goes straight to grey X.

Thanks for the great project... I hope I can get it back to being functional!

rtanaka, 2019-03-06:
hey @ppamidimarri @lolento @acostein - i just got the update and noticed the same thing. can you guys try and post your logs -- i will do the same and try and see what's going on.

ppamidimarri, 2019-03-07:
> hey @ppamidimarri @lolento @acostein - i just got the update and noticed the same thing. can you guys try and post your logs -- i will do the same and try and see what's going on.

I am on a trip out of town, will post my logs when I get home.

lolento, 2019-03-08:
I wish I can help but I'm also traveling and I already repurpose the Pi0W as a pi-hole.

ppamidimarri, 2019-03-10:
@rtanaka I just got back home and I have been trying to find any relevant logs. I don't see anything. The process I used to set up my Pi is non-standard. Here is what I am doing:

1. Start with a clean install of Raspbian Stretch Lite
2. Install nginx, php-fpm and php-mbstring
3. Install and configure rclone, download my website scripts, verify that my website is running
4. Change `/etc/rc.local` to do a filesystem check, run modprobe and prepare some temp files for PHP and nginx to run properly:
```
LOGFILE=/tmp/rc.local.log

function log () {
        echo -n "$( date )" >> "$LOGFILE"
        echo -n ": " >> "$LOGFILE"
        echo "$1" >> "$LOGFILE"
}

log "Running fsck..."
/sbin/fsck /mnt/cam -- -a >> "$LOGFILE" 2>&1 || echo ""
log "Running modprobe..."
/sbin/modprobe g_mass_storage >> "$LOGFILE" 2>&1
log "Preparing temp files..."
/bin/cp /root/.config/rclone/rclone.conf /tmp/rclone.conf >> "$LOGFILE" 2>&1
/bin/chmod 644 /tmp/rclone.conf >> "$LOGFILE" 2>&1
/bin/mkdir /var/log/nginx >> "$LOGFILE" 2>&1
/bin/mkdir -p /tmp/php/sessions >> "$LOGFILE" 2>&1
/bin/chown www-data:pi /tmp/php/sessions >> "$LOGFILE" 2>&1
/bin/mkdir -p /tmp/log/nginx >> "$LOGFILE" 2>&1
log "Starting nginx..."
/usr/sbin/service nginx start >> "$LOGFILE" 2>&1
log "All done"

exit 0
```
5. Download setup-teslausb from this repository, export shell variables `RCLONE_ENABLE`, `RCLONE_DRIVE`, `RCLONE_PATH` and `campercent` and run the setup script 
6. Reboot

This process used to work perfectly before 2019.5.x. Now with a newly formatted Micro SD card, I ran all these steps. It ran fine with no error messages at all. 

The log files I do see now are the ones for my nginx server and they are working just fine. The rc.local log file in /tmp looks normal as well:
```
pi@dpi:~ $ cat /tmp/rc.local.log 
Thu Nov  3 10:17:04 PDT 2016: Running fsck...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
/backingfiles/cam_disk.bin: 5 files, 5/2144943 clusters
Thu Nov  3 10:17:07 PDT 2016: Running modprobe...
Thu Nov  3 10:17:07 PDT 2016: Preparing temp files...
Thu Nov  3 10:17:07 PDT 2016: Starting nginx...
Thu Nov  3 10:17:10 PDT 2016: All done
```

Now when I plug my Pi0W into my Windows 10 computer, even the computer is **not** showing the CAM and MUSIC drives. Those drives showed up fine on the computer the first time it rebooted (after step 5 above), before I tried the Pi0W (without success) in the car with 2019.5.3.

To try to see what's going on, I tried running `sudo modprobe g_mass_storage` over SSH. It ran without any error messages, but still no drives show up on the computer. After that, if I do `mount /mnt/cam` on the Pi, I can `cd /mnt/cam` and `ls` to see that there is a directory called TeslaCam in there. It is empty. `umount /mnt/cam` works without producing any error messages either, as does `sudo modprobe -r g_mass_storage`. 

As far as I can tell, the contents of `/etc/modprobe.d/g_mass_storage.conf` look good:
```
options g_mass_storage file=/backingfiles/music_disk.bin,/backingfiles/cam_disk.bin removable=1,1 ro=0,0 stall=0 iSerialNumber=123456
```
 and `df -H` shows mutable and backingfiles with space allocated to them:
```
pi@dpi:~ $ df -H
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       1.8G  1.3G  434M  74% /
devtmpfs        223M     0  223M   0% /dev
tmpfs           228M     0  228M   0% /dev/shm
tmpfs           228M  6.1M  222M   3% /run
tmpfs           5.3M  4.1k  5.3M   1% /run/lock
tmpfs           228M     0  228M   0% /sys/fs/cgroup
tmpfs           228M   17k  228M   1% /var/log
tmpfs           228M   13k  228M   1% /tmp
tmpfs           228M     0  228M   0% /var/tmp
/dev/mmcblk0p1   46M   23M   23M  51% /boot
/dev/mmcblk0p4   99M  1.6M   90M   2% /mutable
/dev/mmcblk0p3  124G  118G     0 100% /backingfiles
tmpfs            46M     0   46M   0% /run/user/1000
```
Is there anything else I can check to find what the problem is?

rtanaka, 2019-03-10:
@ppamidimarri - thanks for a bunch of information. i've been trying to get to the bottom of it myself with no luck. can you please share your /tmp/archivelog.txt and dmesg contents preferably after it fails to mount (i.e. the "X" shows up for the camera). i've tried re-formatting as well as numerous fsck attempts but i still can't seem to get to the bottom of it.

ppamidimarri, 2019-03-10:
@rtanaka I don't have a file called `/tmp/archivelog.txt` and I think that is because my set up process doesn't install an auto-archiving script. I archive selected files to gdrive using the web interface I built. That was working on previous versions of the car software. Now the car won't see the drive so there are no files to display and select for archiving.

therealmarcone, 2019-03-10:
Finally got 2019.5.15 today, and also got the grey X after the update. The problem seems to be caused by the filesystem image itself though, because when I used 'dd' to make a copy of an existing working sd card and then copied that onto the Pi in place of /backingfiles/cam_disk.bin, it worked again.

While playing with this I also noticed some peculiar behavior with Windows 10:
- right click and eject the music and cam drives
- unplug the pi (I have it on a battery, so it stays powered)
- plug it back in
- Windows will show the drives, but will show them as unformatted / unknown filesystem
- ssh into the pi, "modprobe -r g_mass_storage", the drives will disappear from Windows
- "modprobe g_mass_storage", and the drives will reappear, this time correctly as music/cam drives
- repeat from step 1

Not sure if it behaved like this before, but I had to this once while it was connected to the car as well, before it would see the drives.

ppamidimarri, 2019-03-10:
@rtanaka Here are my dmesg contents from boot (a few hours ago), connected to my Windows 10 laptop, until now: 

```
[Mar 9 15:19] Booting Linux on physical CPU 0x0
[  +0.000000] Linux version 4.14.98+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1200 Tue 
[  +0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[  +0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[  +0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[  +0.000000] Memory policy: Data cache writeback
[  +0.000000] cma: Reserved 8 MiB at 0x1b400000
[  +0.000000] On node 0 totalpages: 114688
[  +0.000000] free_area_init_node: node 0, pgdat c09caaf0, node_mem_map db010000
[  +0.000000]   Normal zone: 1008 pages used for memmap
[  +0.000000]   Normal zone: 0 pages reserved
[  +0.000000]   Normal zone: 114688 pages, LIFO batch:31
[  +0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[  +0.000000] pcpu-alloc: [0] 0 
[  +0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[  +0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B
[  +0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[  +0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[  +0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[  +0.000000] Memory: 435212K/458752K available (6451K kernel code, 589K rwdata, 1992K rodata, 440K init, 673K bss, 15348K reserved, 819
[  +0.000000] Virtual kernel memory layout:
                  vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                  fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                  vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                  lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                  modules : 0xbf000000 - 0xc0000000   (  16 MB)
                    .text : 0xc0008000 - 0xc0655050   (6453 kB)
                    .init : 0xc08d4000 - 0xc0942000   ( 440 kB)
                    .data : 0xc0942000 - 0xc09d54f8   ( 590 kB)
                     .bss : 0xc09dadf0 - 0xc0a8338c   ( 674 kB)
[  +0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[  +0.000000] ftrace: allocating 24128 entries in 71 pages
[  +0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[  +0.000000] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[  +0.000033] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[  +0.000084] bcm2835: system timer (irq = 27)
[  +0.000569] Console: colour dummy device 80x30
[  +0.000026] console [tty1] enabled
[  +0.000035] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[  +0.059513] pid_max: default: 32768 minimum: 301
[  +0.000492] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[  +0.000020] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[  +0.001299] Disabling memory control group subsystem
[  +0.000167] CPU: Testing write buffer coherency: ok
[  +0.001140] Setting up static identity map for 0x8200 - 0x8238
[  +0.001244] devtmpfs: initialized
[  +0.009641] random: get_random_u32 called from bucket_table_alloc+0x88/0x1c4 with crng_init=0
[  +0.001012] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[  +0.000384] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[  +0.000030] futex hash table entries: 256 (order: -1, 3072 bytes)
[  +0.001360] pinctrl core: initialized pinctrl subsystem
[  +0.001344] NET: Registered protocol family 16
[  +0.003087] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[  +0.006679] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[  +0.000017] hw-breakpoint: maximum watchpoint size is 4 bytes.
[  +0.000132] Serial: AMBA PL011 UART driver
[  +0.002947] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[  +0.000721] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[  +0.039236] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[  +0.002305] SCSI subsystem initialized
[  +0.000220] usbcore: registered new interface driver usbfs
[  +0.000095] usbcore: registered new interface driver hub
[  +0.000210] usbcore: registered new device driver usb
[  +0.006736] raspberrypi-firmware soc:firmware: Attached to firmware from 2019-02-12 19:42
[  +0.002132] clocksource: Switched to clocksource timer
[  +0.085898] VFS: Disk quotas dquot_6.6.0
[  +0.000123] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[... 173 lines cut ...]
[  +0.003192] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  +0.000018] g_mass_storage gadget: g_mass_storage ready
[  +0.000014] dwc2 20980000.usb: bound driver g_mass_storage
[Mar 9 16:27] Mass Storage Function, version: 2009/09/11
[  +0.000024] LUN: removable file: (no medium)
[  +0.000219] LUN: removable file: /backingfiles/music_disk.bin
[  +0.007542] LUN: removable file: /backingfiles/cam_disk.bin
[  +0.000027] Number of LUNs=2
[  +0.005013] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  +0.000024] g_mass_storage gadget: g_mass_storage ready
[  +0.000020] dwc2 20980000.usb: bound driver g_mass_storage
[Mar 9 16:30] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
```
[... cut ...]
[  +0.000021] LUN: removable file: (no medium)
[  +0.000156] LUN: removable file: /backingfiles/music_disk.bin
[  +0.000091] LUN: removable file: /backingfiles/cam_disk.bin
[  +0.000009] Number of LUNs=2
[  +0.003197] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  +0.000018] g_mass_storage gadget: g_mass_storage ready
[  +0.000014] dwc2 20980000.usb: bound driver g_mass_storage
```
I am not sure if this is what I should expect on successful completion. Are you able to glean anything from this?

ppamidimarri, 2019-03-10:
@therealmarcone I used to do `sudo modprobe -r g_mass_storage` to make the drives disappear from the car so I could display them in my website, and once all done with archiving to gdrive, I would run `sudo modprobe g_mass_storage` to make them visible to the car. I tested this many times on my Windows 10 laptop and it worked fine, all the way until my car upgraded to 2019.5.3. Now my CAM and MUSIC drives won't even show up on the laptop after I run `sudo modprobe g_mass_storage` from a clean reboot, when then Pi is plugged into the USB drive of my laptop.

ppamidimarri, 2019-03-10:
@rtanaka @therealmarcone I feel so stupid. The reason the drives were not showing on my laptop was that I had the wrong Micro USB port on the Pi connected. The CAM and MUSIC drives now show up properly on my laptop, but the car still shows the grey X after the Pi boots up. 

I will get the dmesg output when the Pi is connected in the car tomorrow.

omarqazi, 2019-03-10:
Dying to get to the bottom of why it is showing up with an X on the Tesla... 

If anyone wants to put their heads together and debug I'm here

therealmarcone, 2019-03-10:
OK, so it appears that the dashcam-side of things wants to see a partitioned device, and not an entire device formatted as fat32. In other words, if you simply
"fallocate -l 16777216K cam_disk.bin" and then "mkfs.vfat cam_disk.bin -F 32 -n CAM" like the setup scripts do, it won't work.
Instead you have to partition the image, then format the partition. At least that's what worked for me. Here's what I did:
- fallocate -l 16777216K cam_disk.bin
- fdisk cam_disk.bin
- create new primary partition: n, p, \<enter>, \<enter>, \<enter>
- set partition type to 'W95 FAT32 (LBA)': t, c, \<enter>
- write partition table: w, \<enter>
- quit fdisk: q, \<enter>

Verify you have a partitioned image with "fdisk -l cam_disk.bin", it should show something like:
```
Device             Boot Start      End  Sectors Size Id Type
cam_disk.bin       2048 33554431 33552384  16G  c W95 FAT32 (LBA)
```
Now since mkfs can only deal with whole devices, you'll need to loopback-mount the partition you just created, so you can create a filesystem on it:

- losetup -o 1048576 loop0 /backingfiles/cam_disk.bin

(the number will vary based on how you partitioned the image file. In my case the partition started at sector 2048, so the offset is 2048*512=1048576)

Create the filesystem and teslacam folder:

- mkfs.vfat /dev/loop0 -F 32 -n CAM
- mount /dev/loop0 /mnt/cam
- mkdir /mnt/cam/TeslaCam
- umount /mnt/cam/
- losetup -d /dev/loop0

Now when you export the image as usb storage, the Tesla will recognize it properly.
For some reason this isn't necessary for the music disk.

therealmarcone, 2019-03-11:
https://github.com/cimryan/teslausb/pull/125 implements what I described above

omarqazi, 2019-03-11:
Hey @therealmarcone -- if we ever meet in real life I owe you a beer or something. I am in California. Let me know. Don't know how you figured that one out.

rtanaka, 2019-03-11:
@therealmarcone - fantastic detective work. i just added the partition to my image and will try on my drive back and report back with the results. it looks promising.

@ppamidimarri - thanks so much for the logs. it definitely helped me narrow down the issue and confirm some of my suspicions.

@cimryan - looks like the @therealmarcone submitted pull request #125 to address this issue. would be good to make sure this is backwards compatible with the older firmware.

therealmarcone, 2019-03-12:
Note that 2019.5.15 also writes the recording files in a different location, and with different names. Instead of creating recent-*.mp4 and saved-*.mp4 files directly in the TeslaCam folder, there are now two folders: TeslaCam/RecentClips and TeslaCam/SavedClips. Every time you hit the save button, it creates another folder under the SavedClips folder, and that folder in turn contains the saved files.
The subfolders are named <date>_<time you pressed save>, and the files themselves also have a different naming pattern, <date>_<time>-front.mp4.
I.e. you might see something like:

```
/mnt/cam/TeslaCam
/mnt/cam/TeslaCam/RecentClips
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-15-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-16-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-17-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-18-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-20-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-25-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-26-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-27-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-28-front.mp4
/mnt/cam/TeslaCam/RecentClips/2019-03-11_17-29-front.mp4
/mnt/cam/TeslaCam/SavedClips
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-12-06
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-12-06/2019-03-11_17-10-front.mp4
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-12-06/2019-03-11_17-11-front.mp4
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-13-54
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-13-54/2019-03-11_17-12-front.mp4
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-13-54/2019-03-11_17-13-front.mp4
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-15-05
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-15-05/2019-03-11_17-13-front.mp4
/mnt/cam/TeslaCam/SavedClips/2019-03-11_17-15-05/2019-03-11_17-14-front.mp4
```

So the various archive-clips.sh scripts need to be updated too, and possibly support both the old and new naming schemes for a while.

ppamidimarri, 2019-03-12:
@therealmarcone Thank you for the fantastic work! I used your updated script to build my drive and it just works. I would have been been able to figure it out. Great job!

therealmarcone, 2019-03-12:
https://github.com/cimryan/teslausb/pull/126 backs up files from both the old and new naming scheme. I only tested CIFS backup.

rtanaka, 2019-03-12:
@therealmarcone - I just updated in 2019.5.15 as well and it looks like it's recording just fine with the changes.

leofaoro, 2019-03-15:
Thanks @therealmarcone now waiting for @cimryan to merge. In the meantime, how can i install your version? thanks

therealmarcone, 2019-03-15:
I haven't tried this so I'm not 100% sure this will work, but I think that if you're using the `teslausb_setup_variables.conf` file to configure your system, you can just set the `REPO` variable's value to `therealmarcone`, and the `BRANCH` variable to `main-dev` (at the bottom of the conf file). This should then make the setup scripts pull the updated versions from my branch.

ppamidimarri, 2019-03-15:
@leofaoro The way I used the code from @therealmarcone was to edit `setup_teslausb` to exit after calling `fix_cmdline_txt_modules_load`, manually getting the correct `create-backingfiles.sh` into `/tmp` using a `wget` command, and then editing `setup_teslausb` again to run the rest of the steps. That worked for me. 

I don't use the auto-archiving feature, so I didn't do this with the updated archiving scripts.

acostein, 2019-03-15:
I was able to reformat the camera partition without rebuilding the Pi from scratch. It took me two days to copy all my music onto the SD card so I wasn't looking forward to doing it all over again.

_This will wipe out your camera partition,_ so copy off any clips you want to keep before doing this.

Here are the steps as I recall them:

1. Power up the Pi, wait a minute for it to boot and connect via WiFi, then SSH into it
2. Become superuser, make filesystem writeable, unmount the camera storage
`sudo -i`
`/root/bin/remountfs_rw`
`umount /mnt/cam`
3. Follow the instructions above by @therealmarcone: https://github.com/cimryan/teslausb/issues/119#issuecomment-471342299 **starting with the fdisk command** (skip the fallocate because you already have a disk image)
4. As referenced in his explanation, add 'offset=...' to /etc/fstab:
`pico /etc/fstab` or `vi /etc/fstab` as you prefer
&nbsp;&nbsp;&nbsp;&nbsp;...and add offset= (use the same number you used for `losetup`) to the cam_disk.bin line:
`/backingfiles/cam_disk.bin /mnt/cam vfat noauto,users,umask=000,offset=1048576 0 0`
5. Mount the new partition and you should be good to go
`mount /mnt/cam`

leofaoro, 2019-03-15:
> I haven't tried this so I'm not 100% sure this will work, but I think that if you're using the `teslausb_setup_variables.conf` file to configure your system, you can just set the `REPO` variable's [...]

Thanks so much I did the headless install and it worked perfectly. Now I'm trying to see if it will send files to my file store.

nate977p, 2019-03-17:
I am on my 5th reload....
I have never used a Pi before and am learning through this process, so I appreciate any help for an absolute newbie. If you want to ignore my silly questions, great I get it :) I ran into the same exact issue as above. I opened a github account and copied the setup-teslausb from cimryan's page and made my own, editing the branch and repo. I had no idea how to do it otherwise; had no idea how to upload from local machine (again complete newbie w/no programming experience!

So that actually seemed to work! But it would not upload... Can anyone help me diagnose? Thank you!

REPO=${REPO:-therealmarcone}
BRANCH=${BRANCH:-main-dev}

jspv, 2019-03-17:
@nate977p Here is my 'fresh build' process.  I have this saved in a files so it's down to just cut/paste.  It allows me to use anyone's repo and branch easily.  Below it is set for my repo (jspv) and branch (main), replace accordingly.  Mine is pulls in the necessary changes from @therealmarcone (thank you!) and adds a minor tweak to handle the new subdirectory structure.  Just set REPO and BRANCH accordingly to pull in whoever's built you want.  

After flashing, setting the wpa_supplicant for WFi, removing any old Pi keys in ~/.ssh/known_hosts,   eject and put in Pi

```
ssh pi@raspberrypi.local
sudo -i
```
customize the next three lines to whatever hostname, and repo, and branch you want to use
```
export TESLAUSB_HOSTNAME=teslausb  
export REPO=jspv 
export BRANCH=main 
```
if you don't use pushover, skip the next three lines
```
export pushover_enabled=true 
export pushover_user_key=_my_pushover_user_key 
export pushover_app_key=_my_pushover_app_key
```
next steps are for rclone setup
`curl -L https://raw.github.com/pageauc/rclone4pi/master/rclone-install.sh | bash`
you can now do `rclone config` or if you know your settings, just write them to the conf file.  Below are my settings for using AWS S3
```
cat > /root/.config/rclone/rclone.conf
[teslas3]
type = s3
provider = AWS
access_key_id =_my_AWS Access Key_
secret_access_key = __my_AWS_Secret_Access_Key_
region = us-east-1
server_side_encryption = AES256
<ctrl-D>
export RCLONE_PATH=_My_AWS_S3_Bucket_Name_
export RCLONE_DRIVE=teslas3
```
almost done, grab the setup and run it
```
wget https://raw.githubusercontent.com/$REPO/teslausb/$BRANCH/setup/pi/setup-teslausb
./setup-teslausb
exit
passwd
```
(change passwd)
`sudo halt`
sometimes the archive doesn't actually get configured, don't know why yet - but if it isn't configured:
(login)
```
mount / -o remount,rw
BRANCH=main
export REPO=jspv
export ARCHIVE_SYSTEM=rclone
export RCLONE_PATH=_My_AWS_S3_Bucket_Name_
export RCLONE_DRIVE=teslas3
export pushover_enabled=true
export pushover_user_key=_my_pushover_user_key 
export pushover_app_key=_my_pushover_app_key
./configure.sh
reboot
```

schmug, 2019-03-19:
Where should I be looking if I'm not seeing any errors yet Tesla is not changing from x status for the dashcam? I tried complete rebuild using #119 and some of the other suggestions here. TeslaCam folder shows when plugging into windows laptop, Google drive looks all good, etc. Haven't seen any errors. 2019.5.15.

jbienz, 2019-03-19:
Has anyone determined what the max _partition_ size is that's supported by Tesla? I've read that FAT-32 can theoretically support up to 16 TB drives, but I'm curious what the largest partition is that's been tested and shown to work with 2019.5.15.

acostein, 2019-03-19:
@jbienz for one data point, my music partition is a bit over 300 GB. takes about 20 mins for the car to scan all the music before I can play anything, but it technically works.

schmug, 2019-03-20:
@jspv Using your suggestions I was able to get almost everything working I think.

How would I go about making sure the Rclone is working? No video has shown up yet even though it is in the SavedClips folder. I've looked around for log files and tried to run the script manually but I'm having trouble.

ppamidimarri, 2019-03-22:
@schmug Two suggestions: 
1. Check that you have a valid rclone.conf file at /root/.config/rclone/rclone.conf
2. Try manually uploading one file with a command line `rclone copyto /mnt/cam/TeslaCam/... gdrive:TeslaCam/...`

raheelsoz, 2019-03-31:
the archive-clip.sh code for rclone leaves the directories intact. This causes the code to try and move the empty directories to the destination archive which results in nothing actually being uploaded but causing the process to delay the USB host connection by several minutes (depending on how many directories are present).
This creates confusion when connected to the car (if you're not reading the logs every time there is a delay), since the MCU can also stop recognizing the drive and require a restart.

I tested adding the following command after the loop to remove all empty directories so the next time the archive is run it will not have to contend with the empty directories.

find /mnt/cam/teslacam/savedclips -type d -empty -delete

Also added the following comment as a reminder why this is there:
# After the file moving loop completes, remove all empty directories, as they cause the arhiveloop to try and move the directories to the destination which takes several minutes and delays the usb host mount causing confusion as the drives are not detected and nothing is actually uploaded.

aberen001, 2019-04-01:
I followed the steps https://github.com/cimryan/teslausb/issues/119#issuecomment-473713815 .. However, when I plug the raspi into my computer, I get a "USB device not recognized" error and I can't see it in My Computer. When I plug it into the Tesla, the dashcam icon never shows up on the screen. Once it boots up, it does blink every second and I can ssh into it and stuff. Not sure where to look.. Any ideas?

barjohn, 2019-04-02:
I followed @jspv's procedure above and when raps is plugged into my computer's USB port the drives show up and I can ping the WiFi IP address just fine.  When I plug it into my car's USB port the Dashcam Icon stay off.  I can ping the device from my computer but the storage is never recognized.  I checked and my boot start was 2048 and I looked at the fstab file and it already had the offset set to the proper value.  Any suggestions welcome.

jspv, 2019-04-02:
> I followed @jspv's procedure above and when raps is plugged into my computer's USB port the drives show up and I can ping the WiFi IP address just fine. When I plug it into my car's USB port the [...]

If you are on 2019.8.x, you may also need to get an updated kernel, via rpi-update.   I'm on 4.19.30 and it's been pretty solid for me.

barjohn, 2019-04-02:
Thanks!  I am updating it now.  I’ll let you know how it went.

John

barjohn, 2019-04-02:
I did the update and still no joy.  I have redone the partition several times and my computer sees it just fine but the Tesla doesn’t.

John

barjohn, 2019-04-03:
> @leofaoro The way I used the code from @therealmarcone was to edit `setup_teslausb` to exit after calling `fix_cmdline_txt_modules_load`, manually getting the correct `create-backingfiles.sh` into [...]

Exactly (step-by-step) how do you do this?

barjohn, 2019-04-03:
@ppamidimarri my question is directed to you above.  Sorry for the confusion as I am trying to set up you web dashcam and I am new at using raspi and LINUX

barjohn, 2019-04-04:
I combined 3 different steps and I get some results:
Step 1, I followed the steps in macron's one step setup and it works to create a working raps (his newest)  Be sure to download  his latest image file. ( [here](https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md) )
Step 2. I followed ppamidimarri's steps ( [here](https://github.com/ppamidimarri/TeslaDashcamWeb) ) up to Configure rclone and used the first part of cimryan's script for clone loading and used the values from ppamidimarri's steps in 2.

When I rebooted the cam and music drives were present.  I then plugged it into my car's USB port and it came up just fine and the dashcam icon showed recording.  He says to access it from the car's browser but Tesla's browser will only go out to the Internet as far as I know and not to a local network IP address.  I used my home computer and connected to it via my browser and I can get the list of files but I cannot upload them, I get a "forbidden" message.  I get the same message if I try and preview the file.

I am assuming the files are somehow locked to a particular user and password so I don't have permission to see the files.

I wanted to look at the file permissions but when I remove the raps from the car and plug it into my computer, the CAM and Music drives do not show.  Running: sudo modprobe g_mass_storage does not make them appear either so I can't look to see what the file protection is.

I am guessing this has to do with my setting up a cis archive in the teslausb_setup_variables.conf file.  However, I really don't know.  All help appreciated.

bassmaster187, 2019-04-04:
I setup a teslausb yesterday (3rd April) and the dashcam icon doesn't appear on my Model S 2019.4.3 - the "USB" Music Icon also doesn't show up. If I plug the raspberry on my windows machine, after approx 1 minute, 2 drives shows up. If i copy a old dashcam video to the raspberry, it will be moved to my windows share. so everything works fine but the tesla dosen't recognize it as an usb thumb drive. 

I updated the kernel via rpi-update without any differences.

barjohn, 2019-04-04:
This thing is madening…using this set of instructions ( https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md <https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md> ) the first few times I was able to get the drives to appear and for the model 3 to recognize the drive.  You do have to wait quite a while after the first boot before they would appear.  Then I was able to modify it and get the web dash cam from here to work following his script ( https://github.com/ppamidimarri/TeslaDashcamWeb <https://github.com/ppamidimarri/TeslaDashcamWeb> ) and files were visibly saved but I could not move them or view them (forbidden message).  I realized I was using both cifs and rcloneso I then went to build without the cifs.  Since then, no folder show, and more issues.  I am thinking it is because I am late and making mistakes so will try again tomorrow.

John

aberen001, 2019-04-04:
I'm going to give this another try today.. Just flashed the firmware, and going to start from scratch

TXRhody, 2019-04-05:
I used jspv's configuration and it is working great. Now to figure out a way to power it long enough to allow all the files to be transferred. It appears the car goes to sleep in my garage before any files are transferred. If I bring the rpi into my house and power it, then all the files get transferred in an hour or so.

barjohn, 2019-04-05:
I am having no joy with marcone’s or any of them so far.  By the way, you can keep the power on by either putting it in sentry mode or turning on climate control.  You can do either or both from your phone app.

John

aberen001, 2019-04-05:
I got Marcones working yesterday using the OneStep method. FINALLY! Are you able to use CIFs or is RClone required in your situation?

barjohn, 2019-04-05:
I had his working a couple of times a few days ago but I wanted to use RCLONE instead of CIFS to give me control over what i decided to upload and keep but for some reason I can’t figure out, the drives no longer appear on my computer after I create the system, even using cifs.

John

lapean111, 2019-04-05:
I am a compete newb, so apologies.  How do I use one of the forks, such as marcones during setup?  The instructions on jspv site still point to the original cimyran for the wget...

barjohn, 2019-04-05:
In the fields for REPO and BRANCH enter marcone’s values, which are marcone and main-dev respectively.  In the wget statement those fields currently have cimryan and master.

John

TXRhody, 2019-04-10:
I am having success keeping power on using this battery pack (https://smile.amazon.com/gp/product/B07BSG7V3J/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1). The battery pack occasionally drains to zero, so I am not sure about how long it will last, but for the time being, it works great.