cimryan/teslausb #120: one step setup broken, troubleshooting steps don't seem to work

marcone2, 2019-03-03:
First troubleshooting step for the one-step headless setup process is "Try sudo -i and then run /etc/rc.local. The scripts are fairly resilient to restarting and not re-running previous steps, and will tell you about progress/failure."

This results in:
root@teslausb:~# /etc/rc.local
My IP address is 192.168.88.149
dos2unix: converting file /root/teslausb_setup_variables.conf to Unix format...
Detecting whether to update wpa_supplicant.conf
Setup doesn't seem to have completed, there is no /root/bin/archiveloop.
Try re-running /root/bin/setup-teslausb (re-downloading if needed),
or export HEADLESS=true and run /etc/rc.local if you want to run automatic setup.

Exporting HEADLESS=true and running /etc/rc.local doesn't work; it just prints the same message again:

root@teslausb:~# export HEADLESS=true
root@teslausb:~# /etc/rc.local
My IP address is 192.168.88.149
dos2unix: converting file /root/teslausb_setup_variables.conf to Unix format...
Detecting whether to update wpa_supplicant.conf
Setup doesn't seem to have completed, there is no /root/bin/archiveloop.
Try re-running /root/bin/setup-teslausb (re-downloading if needed),
or export HEADLESS=true and run /etc/rc.local if you want to run automatic setup.

Running /root/bin/setup-teslausb doesn't work either, because the free space check fails:
(I think this is because the drive has already been partitioned by the setup script at this point, so there is no unpartitioned space left, but the script still expects to find enough unused space to create the backingfiles partition)

root@teslausb:~# /etc/rc.local
My IP address is 192.168.88.149
dos2unix: converting file /root/teslausb_setup_variables.conf to Unix format...
Detecting whether to update wpa_supplicant.conf
Setup doesn't seem to have completed, there is no /root/bin/archiveloop.
Try re-running /root/bin/setup-teslausb (re-downloading if needed),
or export HEADLESS=true and run /etc/rc.local if you want to run automatic setup.
root@teslausb:~# /root/bin/setup-teslausb
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done
Verifying that the requested configuration is valid...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   693  100   693    0     0   1138      0 --:--:-- --:--:-- --:--:--  1141
Downloaded /tmp/verify-configuration.sh ...
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small.

(it's 128GB card, so plenty big)

therealmarcone, 2019-03-03:
The message that /etc/rc.local prints is wrong. Submitted a pull request for correction.

Merckle, 2019-03-05:
> Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
Starting to create backing files partition...
Checking existing partitions...
Modifying partition table for backing files partition...
Error: The location 256087425024 is outside of the device /dev/mmcblk0.
> 

That is what I am getting when I run the setup

mattster98, 2019-03-29:
Getting the "location outside the device" as well.  The root partition is already using the whole as card.. not sure how that's happening.

I'm going to try reimaging the card and creating a 3rd partition so the root can't be expanded into it before booting in the raspi.

My log snippet:
...
Checking existing partitions...
Modifying partition table for backing files partition...
Error: The location 32010928128 is outside of the device /dev/mmcblk0.
root@raspberrypi:~# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 29.8 GiB, 32010928128 bytes, 62521344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x4cd3a954

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    98045    89854 43.9M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 62521343 62423040 29.8G 83 Linux
root@raspberrypi:~#

mattster98, 2019-03-29:
Looks like the cmdline.txt has an initial script to expand the filesystem on first boot, so if you edit that while doing the rest of the first-time-setup stuff in the boot partition, it'll help.  Trying that now.

mattster98, 2019-03-29:
Using guidance here: https://raspberrypi.stackexchange.com/questions/56621/temporarily-disable-expand-filesystem-during-first-boot

I simply deleted the init= argument from /boot/cmdline.txt before first boot and that left the ~2G root partition per the image and left the rest of the device unallocated.  Running setup now - fingers crossed!

mattster98, 2019-03-29:
aaand, this is already in the repository somehow - I guess because of the way I'm doing this (on a chromebook) I'm not getting the benefit.

https://github.com/cimryan/teslausb/blob/master/pi-gen-sources/00-patches/01-cmdline-update.diff

aberen001, 2019-03-29:
Hi! I'm working on this today and have had to start over like 10 times.. Were you able to bypass this error by updating the cmdline.txt file?

mattster98, 2019-03-29:
Yep - just delete init= and everything after it before you boot from the sd card the first time.  I'm still dealing with other issues, but I'm mixing-and-matching this and another repo in order to make it work on the latest Tesla software version.

aberen001, 2019-03-29:
I'll give that a shot. Let me know if you get it working!

aberen001, 2019-03-29:
https://www.reddit.com/r/teslamotors/comments/b67f8c/single_usb_drive_for_both_music_and_cam/

mattster98, 2019-03-29:
Pretty sure that's already implemented with the "campercent" variable - although I had mine set to 100% since I just stream and it still created two partitions - might not have plugged it into the right spot.

The idea is the campercent is the percent of the unallocated portion of the sd card that is allocated for the camera files, and the remainder is music.

aberen001, 2019-03-29:
ah ok, I'll just get through the set up and see if it works then  :)

Thanks brother!

mattster98, 2019-03-29:
Sure thing - it looks like it defaulted to something like 90% for the camera, 10% for music.  Maybe I'll give it a try - I can live without a couple GB of camera file space.

aberen001, 2019-03-29:
Mattster98, it worked!!  "delete init= and everything after it"