cimryan/teslausb #123: Sync?

mg-chen, 2019-03-05:
I've gone through all the steps and now have footage on my rpi. I originally thought the footage would automatically save to Google Drive but my drive is still empty and I haven't received a notification. Am I supposed to do something for the rpi to save the footage to Google Drive?

therealmarcone, 2019-03-05:
Did you manually save the clips (by pressing the button on the screen) ?
By default the archive script only backs up the manually saved clips, not the 1-minute continuously recorded clips.

Did you check /mutable/archiveloop.log ?

rtanaka, 2019-03-06:
Can you check to make sure rclone is configured correctly? You should see something like this if it is correct.

```
root@teslausb:/home/pi# rclone lsd gdrive:
            -1 2018-10-23 00:17:26        -1 TeslaCam
```

mg-chen, 2019-03-06:
@therealmarcone Yea I've been manually pressing the button on the screen. See attached for my logs

[Logs.txt](https://github.com/cimryan/teslausb/files/2933870/Logs.txt)

@rtanaka Below is what I'm getting after running that command but this worked during the setup process and created/displayed my TeslaCam folder. So I'm guessing it stops working after I run './setup-teslausb' in the final steps

`root@teslausb:/home/pi# rclone lsd gdrive:
2019/03/06 01:04:33 NOTICE: Config file "/root/.rclone.conf" not found - using defaults
2019/03/06 01:04:33 Failed to create file system for "gdrive:": didn't find section in config file`

mg-chen, 2019-03-06:
Update- I am now able to receive notifications. I was notified my files were uploaded by Pushover but I still do not see the files in my Google Drive folder.

rtanaka, 2019-03-06:
@mg-chen - oops. sorry it's been a while since i looked at this... can you first confirm the config is set up correctly by running this command?

```
root@teslausb:~# ls -l /root/.config
total 0
lrwxrwxrwx 1 root root 23 Oct 31 10:25 rclone -> /mutable/configs/rclone
root@teslausb:~# 
```

and then run this command if the above works.

```
root@teslausb:~/bin# rclone --config /root/.config/rclone/rclone.conf lsd gdrive:
          -1 2018-10-23 00:17:26        -1 TeslaCam
root@teslausb:~/bin# 
```

mg-chen, 2019-03-06:
@rtanaka the first command worked but not the second one. See below

[image]

rtanaka, 2019-03-06:
@mg-chen - so it looks like your rclone drive configuration is not correct. you'll have to re-do it.

create the file as root with this command:

```
rclone --config  /root/.config/rclone/rclone.conf config
```

1. select "n" for new remotes
1. use "gdrive" for name
1. select google drive for type (should be 12)
1. leave client_id and client_secret blank
1. use drive.file as the scope (option 3)
1. leave root_folder_id and service_account_file blank
1. skip advanced config (n)
1. select "N" for autoconfig
1. copy the link to create a token
1. copy that verification code
1. say no to team drive
1. y to confirm

try the command again and hopefully it'll work

mg-chen, 2019-03-07:
@rtanaka I'm going through the setup process but I'm receiving this notice. I was thinking I can just create the file but I'm not sure if there is supposed to be code in it to begin with.

[image]

rtanaka, 2019-03-07:
@mg-chen - not sure exactly what happened but it's clear the config file got removed, deleted or was never created. without it, there's no way it will work.

mg-chen, 2019-03-08:
@rtanaka That's a bummer. Thank for your help though!

rtanaka, 2019-03-08:
no problem @mg-chen - were you able to get it to work after you restored the config?

mg-chen, 2019-03-12:
@rtanaka Sorry for the late response, I've been pretty busy the last few days so I haven't been able to work on it as much. When I was last working on it my rclone.conf file would get removed by running `./ setup-teslausb` even when I manually created the file. 

I believe my next option would be to create the rclone.conf file in the .config directory and find out how to change the path to /root/.config/rclone.conf instead of /root/.config/rclone/rclone.conf

rtanaka, 2019-03-13:
the set up scrips do not delete the rclone.conf file, it only just moves it.

you can see how to specify the config file location in one of my previous comments above. if the file is not there, it will create it after you successfully complete the config process.

mg-chen, 2019-03-14:
@rtanaka Where does the script move rclone.conf to? Because I'm still getting the following in my logs.
[image]

rtanaka, 2019-03-15:
the command that moves the configs is part of configure-archive.sh and does the following:

```
    mv /root/.config/rclone /mutable/configs
    ln -s /mutable/configs/rclone /root/.config/rclone
```

schmug, 2019-03-20:
> Did you manually save the clips (by pressing the button on the screen) ?
> By default the archive script only backs up the manually saved clips, not the 1-minute continuously recorded clips.
> 
> Did you check /mutable/archiveloop.log ?

I think I am having this issue as well but I do not have a /mutable/archiveloop.log

rclone lsd tesla3: works when I am mounted as read/write, I have files in TeslaCam and SavedClips but I don't believe the scripts are running.

therealmarcone, 2019-03-20:
@schmug did you do headless setup from the prebuilt images from rtgoodwin by any chance?
I found the included sample conf file is missing the ```export ARCHIVE_SYSTEM=...``` line, and without that it won't install the proper scripts for your chosen backup method.

schmug, 2019-03-20:
> @schmug did you do headless setup from the prebuilt images from rtgoodwin by any chance?
> I found the included sample conf file is missing the `export ARCHIVE_SYSTEM=...` line, and without that it won't install the proper scripts for your chosen backup method.

I may have crippled myself but I have rebuilt so many times I cannot remember my last steps, should have documented. I tried to follow [jspv's advice](https://github.com/cimryan/teslausb/issues/119#issuecomment-473713815) and I think I have everything working but no rclone script running

I will look there. I believe #128 is happening for me as well but running `sudo modprobe -r g_mass_storage and sudo modprobe g_mass_storage` seems to flip from Dashcam icon with gray dot to red dot (and working)

Rykemapo, 2019-03-20:
If these pull requests don't get merged soon, we may need to make @therealmarcone fork the new master.

jspv, 2019-03-20:
@schmug Did you try running configure.sh to configure the scripts (last section on my advice)  I've had times that I needed to run that manually.  

Also if you are using s3, you may want to double check your s3 policies to ensure you're allowing write.  Trying a a manual 'rclone copy' to your bucket would be a good test.

schmug, 2019-03-20:
@jspv thanks for the reply. I had in one iteration but I think I hadn't since I got it working better. I manually edited configure.sh since it had reference to cimryan to jspv and just ran it and rebooted but it isn't back yet so I'm headed to my car to take a peek and see what's wrong. Also am I reading it right that it will only occur on power on?

jspv, 2019-03-20:
You don’t need to edit configure.sh to change REPO or BRANCH, the hard coded cimryan only takes effect if you aren’t setting the BRANCH and REPO environment variables. If you set the variables, it will pull the script from the repo you specify. 

Only need to run configure.sh once, it downloads the scrips to automate the device.

schmug, 2019-03-20:
@jspv makes sense I think I was just being dense because it took me around 6 tries to get working haha. Does rclone only trigger automatically on boot or is it on wireless connect?

jspv, 2019-03-20:
When you boot, archiveloop gets automatically run and keeps running forever. Archiveloop will check for a network connection, if it finds one it will sync the files. When done, it will turn the device into a usb drive and wait for a network disconnect and a later reconnect, at which time it will sync the new files and repeat the process.  

You can fake a connection or disconnect by touching /tmp/archive_is_reachable and /tmp/archive_is_unreachable respectively for testing.

schmug, 2019-03-21:
@jspv so I can say that [your advice](https://github.com/cimryan/teslausb/issues/119#issuecomment-473713815) also worked for me on 2019.5.15. thanks!

rochec, 2019-06-15:
Has anyone been able to find a solution for the config issues? I'm having the same issue that @mg-chen was having.

Should there be a .config directory? The contents of root are bin  rpi-sync  teslausb.conf

I've run through the config setup repeatedly and this happens each time.