cimryan/teslausb #127: Interactive website to view videos and choose them for archiving

ppamidimarri, 2019-03-15:
I have setup my Pi to serve a website that shows me all the videos that are saved in the TeslaCam folder (both recent and saved clips), allowing them to be previewed, deleted, renamed, or uploaded to my Google Drive. I prefer this over auto-archiving all saved clips the next time the Pi connects to my home WiFi. An unexpected benefit of this approach is that when Tesla changed its layout and file naming schemes with 2019.5.x, the website just works without the need for any changes. 

I built this using nginx, PHP and [ResponsiveFileManager](https://www.responsivefilemanager.com/). 

I am happy to share my code here, but I don't know how to go about that. The setup procedure I use is to first setup the web server and my scripts on the Pi and then run setup_teslausb from this repository to make the Pi read-only. I am not good with shell scripting, so I don't know how to integrate this into the excellent headless setup procedure developed here. If there is any interest in integrating the website feature into this project, please let me know!

chilman408, 2019-03-26:
Looks great, does it work if I install this over the Teslausb setup?

ppamidimarri, 2019-03-26:
@chilman408 Yes, it works with teslausb. There are two ways you can get it working: 
1. Set up a new Pi following [the instructions at my repository](https://github.com/ppamidimarri/TeslaDashcamWeb). In this approach, you first set up the website on the Pi and then create the USB drives using teslausb.
2. Install the website on an existing teslausb Pi. I haven't tested this, but you can probably remount `/` as read-write and then install the website and it will probably work.

ppamidimarri, 2019-03-29:
@lolento I haven't tried it myself this way, but it should work. The folders touched are /etc, /var/www and /home/pi.

barjohn, 2019-04-03:
I was able to follow your instructions in your repository up to setting up the USB drives.  I did have a couple of problems.  One was an error which you appear to have corrected in this line: "sudo nano /etc/nginx/nginx.conf"  You had nginx.confg.  That one was an easy guess to fix.  The update sudoers was a challenge as I wasn't sure how to create the file that did not exist with VISUDO but I think I finally figured it out.  The next problem I had was using rclone as you didn't make it clear that one had to load it first.  Now, where I am having difficulty and it would be great to have a clean set of instruction in your instructions is in setting up the USB drives.  How to use macron's teslausb-setup script and modify it to exit after the fix_cmdline_txt_modules_load call and knowing which was the correct create-backing files.sh file and how to get it into raps and then restart the script from after the exit point seems difficult to me as a noob.  Any help you can provide would be welcome.  Having gotten this far after many hours I don't want to screw it up and have to start over.