cimryan/teslausb #131: USB ports disabled with 2019.8.2

drpotters, 2019-03-24:
I followed the discussion in https://github.com/cimryan/teslausb/issues/119 and was able to get this working again w/ the newly partitioned cam disk image, until I updated to 2019.8.2 5 days ago.. Since then, I can see the Pi starting up and then it goes dark (no power) with the USB port shutdown. The only way I can get the port active again is to restart the MCU, and I have the same problem if I plug in to the other front port..  It looks to me like the vehicle is detecting this as an attack/problem and disables the port.

firedfly, 2019-03-24:
I was having the same issue with the power going out to the USB ports.  I resolved the problem by powering the pi via the cigarette outlet.  Haven't experienced the issue again since I made that change a few days ago.

drpotters, 2019-03-24:
Providing another source for power, as @firedfly suggests, appears to work around the problem and my teslausb is once again working. Since the USB ports can deliver up to 2 amps at 5v I'm surprised the car is turning them off while driving. From what I can tell, a Raspberry Pi Zero W should only use 120-230mA of power, and I'm willing to bet there's something that can be done with the teslausb to have the vehicle continue to deliver power. I'm going to contact Tesla to see what, if anything, they might suggest.

Anyway, I hope this will help others to more readily identify what's happening.

lolento, 2019-03-29:
I'm on 8.3 and I dont have this issue, just fyi

ScrawnyB, 2019-03-30:
I'm on a 'different version' beyond 2019.8.x *cough* but have had the same issue. I've noticed a USB hub (that has a Power LED) in front of the Pi seems to help keep it on better. I'm still having some issues, but I think I need to apply the kernel fix everyone is talking about.

japhule, 2019-03-30:
Having the same issue as @ScrawnyB, the kernel fix didn't fix it for me either. I went a whole day without issue with sentry mode on for several hours. The issue cropped up again though on day 2 so it does not appear to be fixed for me. I'm also powering the rpi with the 12v.

neondemon, 2019-04-01:
Im also in the beta program, and having the same issue @ScrawnyB . External power resolves it entirely, but is not a good solution imo. 
Think we should just report to tesla? if this is a power over usb issue, can it be solved another way?

jspv, 2019-04-01:
Some discussion of this here, including an email response from Jeda who is also looking at the problem https://teslaownersonline.com/threads/usb-drives-dying-early-deaths-when-used-for-tesla-dash-cam-videos.11828/page-3#post-221333

> Our team has conducted extensive testing using the Jeda Wireless Pad and many other accessories that use USB cables. What we have seen is that when those accessories make use of the data wires inside a USB cable, the Tesla Model 3 resets or interrupts the power connection. This happens with many wireless chargers, battery packs and other car accessories.
> ...
> One way or another, the solution is to replace the existing USB cable for a pair of USB cables that are charging only, which means they don't use the data wires. 

Note: I'm on 2019.8.3 and I am **not** having the issue, although I fully expected to when I updated last week.  My rpi stays happily connected, even with long Sentry modes, and has been updating fine when I get home.  I **am** using a USB splitter, the one that came with the Nomad charger, with the 'data' side used for the rpi and the power side to the Nomad.  Per the post from Jeda, it appears that that using the splitter may be the reason why I have not experienced this.

leofaoro, 2019-04-02:
My system is now working (with jsvp's fork), but I had to connect the pi to a USB hub which in turn is connected to the car. I also used rpi-update but not sure that fixed anything.

x-cimo, 2019-04-02:
Leofaoro, can you share the USB hub model?

Thanks

leofaoro, 2019-04-03:
> Leofaoro, can you share the USB hub model?
> 
> Thanks

Sure this is what I have
Sabrent 4-Port USB 3.0 Hub with Individual LED Lit Power Switches, Included 5V/2.5A Power Adapter (HB-UMP3) https://www.amazon.com/dp/B00TPMEOYM/ref=cm_sw_r_cp_tai_-YaPCb8FTJNCP