cimryan/teslausb #132: Boot loop when connected to car

mattster98, 2019-03-29:
My guess is this isn't software-related, but wanted to at least solicit input on this issue.  I got updated to 2019.8.3 and had to update to get this to continue to work (re #119).   I had previously rolled-my-own from the reddit thread so I started with a clean wipe.

After a lot of weird issues that were mostly my fault, I got the image working such that when plugged into my laptop, the pi boots, and presents two usb drives to the host.

However, when I connect it to the car, as best I can tell, the pi gets into a boot loop - the led is solid light for a bit, flashing like typical boot, goes off for a bit, then solid, then flashing.  SSH will connect then hang when it reboots after just a few seconds - not long enough to see what's going on.

Back to the laptop and it works fine.

I'm not using a hub or anything - connecting via a cable I've been using exclusively for this - and have tried both USB ports in my X.

Plain-old USB flash drive works fine.

laudbrian, 2019-03-29:
In case this helps, I just got mine up this morning, I'm using CIFS. 
Make sure the kernel was updated, there is a bug with fsck function. 
In step 2 after you ssh into the Pi:
sudo rpi-update
Also, at the end of setting up everything I had to provide additional juice to the rPi0 with a USB power bank before it started transferring video files to my server, ordered a battery hat to replace this later.

mattster98, 2019-03-29:
Thanks for the suggestion.  I updated and tried again with the same result, rebooting.  Added a second cable for power and it seemed to be OK.  Just to completely rule out the cable I was previously using, I used the new cable as the sole cable AND IT KEPT WORKING, DAMNIT!

So - tldr - bad cable..  after working fine for weeks and weeks, is no longer fine.  I f'n hate micro usb cables.

mattster98, 2019-04-01:
OK - not reliably working still.  Definitely seems to be power-related.. I don't understand why this was working fine for months and now it's not just because I updated.  Going to look into power-limiting the Pi somehow.

mattster98, 2019-04-01:
Tried limiting cores, limiting CPU freq (400mhz), and disabling HDMI interface, but haven't been able to get it to reliably boot and stay running while powered by the single USB interface.  Connecting to both seems to work, but I don't understand why that's now necessary.

I can't tell what specific part of the boot is causing it to run out of power, but it's after the network comes up.  Perhaps there's some power settings that could be adjusted.

Is anyone running the latest version and NOT having this issue??

mattster98, 2019-04-01:
The further I dig into this, I think this is a dupe of #128

closed 2019-04-01 by mattster98