cimryan/teslausb #135: Nothing backing up

macbentosh, 2019-05-07:
Setup went ok, however no files are backing up to the network share. Where should I start looking?

marcone, 2019-05-07:
Does the dashcam icon show on the car's screen? If not, setup didn't actually complete successfully.
Did you press the icon to save something? If not, there's nothing to back up, since teslausb only backs up saved clips currently.
Did you leave and re-enter wifi range, or reboot the rpi? If not, teslausb won't start the backup operation.

macbentosh, 2019-05-07:
Ok might be the first one testing this then. I have a non AP 2.5 car so I do have have Dashcam only sentry. I am also powering the pi with constant power on the power port and usb on the other port. How does it kick off automatically usually?

marcone, 2019-05-07:
Not sure how Sentry-only works, but on my car the files saved while Sentry Mode are saved the same as manually saved dashcam files, i.e. in folders under TeslaCam/SavedFiles/
So first check that something actually got saved there: take the rpi out of the car and plug it into a PC, wait for it to boot up and publish the drive(s), and browse around on the camera drive.

The way the automatic backup process normally works is that once the car returns within wifi range, the drive(s) will be disconnected from the car (same as if you had unplugged them), mounted on the rpi, then the backup script backs up the saved files using the selected method, then reconnects the drives to the car.

macbentosh, 2019-05-10:
Is there a log to look at somewhere?

marcone, 2019-05-10:
Setup log is /boot/teslausb-headless-setup.log (if you ran the headless setup), the archiveloop script log is in /mutable

miles267, 2019-10-04:
Finally have teslausb running with v10. Connectivity to WiFi without issue. default CIFS 1.0 setting. Can access the windows server 2012 file share successfully from RP0. How can I troubleshoot why nothing is uploading to the file share?