cimryan/teslausb #137: Followed all steps, no dashcam icon in Tesla

shaktipatel, 2019-05-07:
No dashcam icon in my 2018 model s. Followed all the steps successfully and rebooted as teslausb.local. I assume to plug in the pi in the data/power then to the tesla, instead of just power. I tried both and still no dashcam icon after minutes.

bassmaster187, 2019-05-07:
I've got the same problem, but it shows up on my PC after a minute or two.

marcone, 2019-05-07:
The current cimryan/teslausb does not work with Tesla firmware 2019.x, and while there is a PR to fix that, it looks like this project is not maintained anymore, so the PR hasn't been merged.

Try https://github.com/marcone/teslausb instead.

mattblumberg, 2019-05-22:
Seems like there is an issue... It hangs here:
Mon  6 May 01:08:52 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:08:52 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Mon  6 May 01:08:52 BST 2019 : Rebooting...
Mon  6 May 01:09:18 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon  6 May 01:09:18 BST 2019 : Grabbing main setup file.
Tue 21 May 06:17:16 BST 2019 : Detecting whether to update wpa_supplicant.conf
Tue 21 May 06:17:16 BST 2019 : Grabbing main setup file.

Hangs at 'grabbing main setup file' forever...