cimryan/teslausb #142: May I use Raspberry Pi 3+ to do this?

ffllc, 2019-06-18:
I had Raspberry Pi 3+ ,so may I use Raspberry Pi 3+ to do this?

hilli, 2019-06-21:
Nope. Raspberry Pi Zero only. The other Pi's have a USB hub on them, which ruins the USB-To-Go.

ffllc, 2019-06-22:
copy that. thanks.

rreynier, 2019-06-27:
Is there any ways to make this possible on other raspberry's? Using some other approach?

hilli, 2019-07-08:
@rreynier, not on the Pi's. But I just found this thing (Alas, I haven't tried it):

https://www.olimex.com/Products/OLinuXino/Home-Server/Pioneer-FreedomBox-HSK/open-source-hardware

It is an ARM7 CPU, 1GB RAM, has USB On-The-Go support _and_ 2 USB 2.0 ports. No wifi though, but you can use a USB WiFi dongle for that. I also like the build in battery that acts a an UPS - Handy for when the Tesla shuts off the power, allowing for a graceful shutdown of the thing.

closed 2019-08-13 by ffllc