cimryan/teslausb #149: Error when running ./setup-teslausb script

slinnings, 2019-07-09:
When running the .setup-teslausb, I get the below error. Please help.

Thanks.

root@raspberrypi:~# ./setup-teslausb
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done        
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.

SemoTech, 2019-07-10:
@slinnings I had a lot of issues with running the script myself. What worked for me was re-flashing the microSD card with the latest image from here, copying the custom config file over, and rebooting.  Worked like a charm after that. Hope this help you as well.

slinnings, 2019-07-10:
@SemoTech , what do you mean with the "latest image from here"? I downloaded the OS to the SD card following the instructions in the readme file.

SemoTech, 2019-07-10:
@slinnings I was referring to the image referenced in the "One-step setup" as it's a turn-key setup:  https://github.com/cimryan/teslausb/blob/master/doc/OneStepSetup.md

endurance1968, 2019-07-15:
Rasbian Buster was just released - to avoid the error "InRelease' changed its 'Suite' value from 'testing' to 'stable'" execute   "sudo apt update --allow-releaseinfo-change" and then sudo apt upgrade. After this also the script will run.