cimryan/teslausb #150: Does not work with new rasbian images due to auto root expansion

endurance1968, 2019-07-15:
Since new rasbian images autoexpand after first boot there is no space left on the device for new partitions. Workaround is to disable autoexpand in cmdline.txt prior tro first boot.

justinporterfield, 2019-08-17:
After you flash the SD card but before you put it into the Pi edit the `cmdline.txt` file and remove the   `init=/usr/lib/raspi-config/init_resize.sh` text at the end of the file.

nowjon, 2021-06-13:
I don't see the line above mentioned in the `cmdline.txt`
This is what's in it: 
```dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=3f98dc8b-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_ether modules-load=dwc2,g_ether quiet```

nowjon, 2021-06-14:
Used this image instead and didn't have the issue: https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/