cimryan/teslausb #151: Error running on raspbian buster -  The location is outside of the device /dev/mmcblk0

choeecs27, 2019-07-19:
Error when running ./setup-teslausb: The location is outside of the device /dev/mmcblk0.  Any idea?

Downloaded /root/configure.sh ...
~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
Starting to create backing files partition...
Checking existing partitions...
Modifying partition table for backing files partition...
Error: The location 32395755520 is outside of the device /dev/mmcblk0.

divahno, 2019-07-28:
the same

divahno, 2019-07-28:
Problem was that I had an initial boot before applying patch for autoresize. Once I re-flashed SD and patched `cmdline.txt` correctly (before boot), then `./setup-teslausb.sh` run without an error.

justinporterfield, 2019-08-17:
Specifically remove the `init=/usr/lib/raspi-config/init_resize.sh` text at the end of the `cmdline.txt` file.

oliverkurth, 2019-08-24:
I am getting the same error. How does `cmdline.txt` have to look? How is this even related to the error?
```
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
Starting to create backing files partition...
Checking existing partitions...
Modifying partition table for backing files partition...
Error: The location 7948206080 is outside of the device /dev/mmcblk0.
```
```
# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
```

oliverkurth, 2019-08-24:
This happened (in my case at least) because the whole disk was already allocated for the /boot and the root partitions, so the script tried to start the new partitions beyond the end of the last one.

tills13, 2019-11-03:
👍 Thanks @justinporterfield / @divahno. You must edit the `cmdline.txt` _before_ first boot, removing the `init` part at the end.