cimryan/teslausb #158: Issue: outside of the device /dev/mmcblk0.

roastduckkiller, 2019-09-24:
When running ./setup-teslausb, I got following error message, how could I troubleshoot this problem.
Thanks

----------------------------------------------
Updating package index files...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Verifying that the requested configuration is valid...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   693  100   693    0     0    813      0 --:--:-- --:--:-- --:--:--   814
Downloaded /tmp/verify-configuration.sh ...
Verifying that there is sufficient space available on the MicroSD card...
There is sufficient space available.
Downloading additional setup scripts.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2134  100  2134    0     0   2350      0 --:--:-- --:--:-- --:--:--  2350
Downloaded /tmp/create-backingfiles-partition.sh ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1474  100  1474    0     0   1904      0 --:--:-- --:--:-- --:--:--  1904
Downloaded /tmp/create-backingfiles.sh ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2194  100  2194    0     0   2952      0 --:--:-- --:--:-- --:--:--  2952
Downloaded /tmp/make-root-fs-readonly.sh ...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  5660  100  5660    0     0   6247      0 --:--:-- --:--:-- --:--:--  6240
Downloaded /root/configure.sh ...
~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
Starting to create backing files partition...
Checking existing partitions...
Modifying partition table for backing files partition...
Error: The location 32105299968 is outside of the device /dev/mmcblk0.

jgimbel, 2019-10-05:
I had the same issue with `export campercent=100`. 
I installed Raspbian and set `export campercent=90`. It was able to make both partitions properly that time. Maybe the script is trying to create a partition that is 0 bytes long? I couldn't quite tell which partition it failed to make.

maynardpa, 2020-01-06:
Tried a 32GB and 128GB card, in both cases, I am unable to partition this properly by letting the script do what it should. I have tried multiple values for campercent, 60%, 80%, 90% and 100% without success. This is a showstopper for me as the Teslacam partition is unusable.

ryanm7780, 2020-01-30:
Same here on a 32GB drive.  Bummed.  This is a really cool idea.  

> Tried a 32GB and 128GB card, in both cases, I am unable to partition this properly by letting the script do what it should. I have tried multiple values for campercent, 60%, 80%, 90% and 100% without success. This is a showstopper for me as the Teslacam partition is unusable.

maynardpa, 2020-02-01:
Hi, I have eventually figured out the root cause of the issue on my end. In etcher and similar apps there is a setting to auto expand the last partition. This needs to be turned off to work properly with the partitioning.

Cheers

(Sent from mobile)

Pier-André Maynard