cimryan/teslausb #164: This project is frustrating me...

bojennett, 2019-11-07:
So, I had the thing die on me and went and rebuilt it.

I have it just inside my house now (not yet in my Tesla).  I can SSH into it.  I can go to /backingfiles and see that there is a cam_disk.bin and a music_disk.bin, even though there should be no freaking music disk as I set up cam disk to be 100%.

I have it running while plugged in via USB to my Mac.  No drive appears.

In the /boot directory, there is a **WIFI_ENABLED** file that got created and a **TESLAUSB_SETUP_STARTED** file (both with 0 bytes)

This is the log **tesla-usb-setup.log** file

> Sun 22 Sep 22:55:13 BST 2019 : Detecting whether to update wpa_supplicant.conf
> Sun 22 Sep 22:55:13 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
> Sun 22 Sep 22:55:14 BST 2019 : Rebooting...
> Sun 22 Sep 22:55:15 BST 2019 : Grabbing main setup file.
> Sun 22 Sep 22:55:45 BST 2019 : Detecting whether to update wpa_supplicant.conf
> Sun 22 Sep 22:55:45 BST 2019 : Grabbing main setup file.
> Sun 22 Sep 22:55:46 BST 2019 : Starting setup.
> Sun 22 Sep 22:55:47 BST 2019 : adding country code to wpa_supplicant.conf
> Sun 22 Sep 22:55:47 BST 2019 : rebooting for changes to take effect
> Sun 22 Sep 22:56:19 BST 2019 : Detecting whether to update wpa_supplicant.conf
> Sun 22 Sep 22:56:19 BST 2019 : Starting setup.
> Sun 22 Sep 22:56:19 BST 2019 : reading config from /root/teslausb_setup_variables.conf

And that's it.  

Keep feeling like it's a "you get what you paid for" project.  It's free, so there ya go...

alectrocute, 2019-12-08:
Try setting the disk variable to 80 instead of 100. I have a feeling it’s failing due to storage constraints.

ulrichanthony, 2020-02-12:
I had a similar issue where the wifi was set to enabled but not connecting. Have you confirmed your wifi is active? https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point/check-ethernet-and-wifi